davidepatti/noxim #1: Segmentation fault (core dumped) encountered on running "noxim object" after "make"

AntaraG, 2015-08-03:
I have been trying to run noxim simulator for the first time after installation but it shows segmentation fault--
$ ./noxim 

        SystemC 2.3.1-Accellera --- Jul 27 2015 13:51:54
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

Loading configuration from file config.yaml
Segmentation fault (core dumped)

Also, I get the following warnings during "make"--
../src/Channel.cpp: In member function ‘void Channel::powerManager(int, tlm::tlm_generic_payload&)’:
../src/Channel.cpp:65:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      if (i!=hub_dst_index) 
             ^

../src/Hub.cpp:42:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (int i=0;i<rxChannels.size();i++)
                                 ^
../src/Hub.cpp:69:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (int i=0;i<rxChannels.size();i++)
                                 ^
../src/Hub.cpp: In member function ‘void Hub::perCycleUpdate()’:
../src/Hub.cpp:88:36: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i=0;i<txChannels.size();i++)
                                    ^
../src/Hub.cpp: In member function ‘void Hub::updateRxPower()’:
../src/Hub.cpp:115:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (int i=0;i<rxChannels.size();i++)
                                 ^

Help me with the debugging details please.

smonteleone, 2015-08-05:
Could you please send me the config.yaml you used with this noxim simulation? You can find my email in the doc/AUTHORS.txt file. Also, any other detail would be helpful.

AntaraG, 2015-08-05:
Mailed. Thank you.

SyedAbubakar, 2015-10-03:
I am also having this problem now on dell lettitude E6430 3rd gen intel core i5 processor. But on other system of HP core i3 wnd gen no error is coming. Please help me to resolve this problem on my system. Thank you.

smonteleone, 2015-10-09:
Hi all, we think we solved the problem. Please update noxim:
From the terminal, cd to noxim/bin directory and type:

	git pull
	make clean 
	make

Now it's possible to test with:

	./noxim -config ../config_examples/default_config.yaml

mamadymadan, 2015-10-10:
not work :-(

smonteleone, 2015-10-10:
Hi, 
could you please do the following:
In noxim/bin/Makefile at line 25, change:

	DEBUG    := # -g -DDEBUG

in

	DEBUG    := -g -DDEBUG

Then save and execute:

	make clean
	make
	gdb noxim

... and from the prompt of gdb type the following gdb command:

	run -config ../config_examples/default_config.yaml

and then another gdb command

	bt

This will give again the error but with a more detailed description. Please send it back to us. Thank you in advance for your collaboration

mamadymadan, 2015-10-10:
root@ubuntu:/home/mohammad/noxim/bin# gdb noxim
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html
>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from noxim...done.
(gdb) run -config ../config_examples/default_config.yaml
Starting program: /home/mohammad/noxim/bin/noxim -config
../config_examples/default_config.yaml
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.
so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

        SystemC 2.3.1-Accellera --- Aug  8 2015 00:43:14
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

        Noxim - the NoC Simulator
        (C) University of Catania

Loading configuration from file ../config_examples/default_config.yaml

Program received signal SIGSEGV, Segmentation fault.
YAML::detail::node_data::get<std::string> (this=0x6cc050, key=...,
pMemory=...)
    at /usr/local/include/yaml-cpp/node/detail/node.h:29
29      NodeType::value type() const { return m_pRef->type(); }
(gdb) bt
#0  YAML::detail::node_data::get<std::string> (this=0x6cc050, key=...,
    pMemory=...) at /usr/local/include/yaml-cpp/node/detail/node.h:29
#1  0x0000000000422a02 in get<std::basic_string<char> > (pMemory=...,
key=...,
    this=<optimized out>)
    at /usr/local/include/yaml-cpp/node/detail/node_ref.h:67
#2  YAML::detail::node::get<std::string> (this=0x6cbc20, key=...,
pMemory=...)
    at /usr/local/include/yaml-cpp/node/detail/node.h:130
#3  0x0000000000422bc2 in YAML::Node::operator[]<char [13]> (
    this=this@entry=0x7fffffffd1d0, key=...)
    at /usr/local/include/yaml-cpp/node/impl.h:390
#4  0x0000000000418627 in loadConfiguration ()
    at ../src/ConfigurationManager.cpp:22
#5  0x000000000041b29b in configure (arg_num=arg_num@entry=3,
    arg_vet=arg_vet@entry=0x6c7140) at ../src/ConfigurationManager.cpp:419
#6  0x00000000004394c1 in sc_main (arg_num=3, arg_vet=0x6c7140)
    at ../src/Main.cpp:46
#7  0x00007ffff7ae1674 in sc_elab_and_sim () from /usr/lib/
libsystemc-2.3.1.so
#8  0x00007ffff6c1fec5 in __libc_start_main (main=0x40ae50 <main@plt>,
argc=3,
    argv=0x7fffffffe5e8, init=<optimized out>, fini=<optimized out>,
    rtld_fini=<optimized out>, stack_end=0x7fffffffe5d8) at libc-start.c:287
#9  0x000000000040ee30 in _start ()

-----------------------------------------------------------------
thanks a lot

mamadymadan, 2015-10-10:
now i change YAML folder in Makefile.
it work without error!
it work without error in debug mode.
I test it in two ubuntu system

smonteleone, 2015-10-10:
It seem you have more than one yaml-cpp lib in your system, and the Makefile is linking the wrong one (please look at https://github.com/jbeder/yaml-cpp/issues/299). For this reason changing YAML path in Makefile works for you.

smonteleone, 2015-10-12:
So, to whom still has the same problem reported by @mamadymadan: Please, check whether you have the folder /usr/local/include/yaml-cpp. In the case you have it, modify the noxim/bin/Makefile changing

	YAML    := /usr
in

	YAML    := /usr/local
otherwise point to your YAML path. Then

	make clean 
	make
It should work also without activating the debug flag

closed 2015-10-12 by smonteleone