davidepatti/noxim #2: No yam cofiguration found on Ubunto 14.0.2 LTS

SyedAbubakar, 2015-09-23:
I run the command you provided but it is not working for me and my fellow too. The same error come on our pc's. Help us to resolve this problem.
When i run noxim following message comes.

SystemC 2.3.1-Accellera --- Sep 23 2015 11:33:47
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

No YAML configuration file found!
 Use -config to load examples from config_examples folder

smonteleone, 2015-09-23:
Hi, I'm not sure I can reproduce the problem.
Initially I thought it was a path related problem but for me both the following commands work:
```
noxim/bin$ ./noxim -config ../config_examples/default_config.yaml 
$ noxim/bin/noxim -config noxim/config_examples/default_config.yaml 
``` 
May you give more details? Please consider also updating noxim to the latest version.

SyedAbubakar, 2015-10-01:
Thank you very much. Your guidance helped for me. :+1: 
I also need your help in using 3D NOC noxim as it is also not working.
I followed this directory.
https://github.com/habedi/SystemC-AccessNoxim/blob/master/README.md
The genuine source of this tool is:
http://access.ee.ntu.edu.tw/noxim/index.html
Kindly help me in resolving issue. The problem occurs in making access noxim. That gives errors related to:

""  ./NoximMain.o:NoximMain.cpp:(.text._ZN8NoximNoCC1EN7sc_core14sc_module_nameE[NoximNoC::NoximNoC(sc_core::sc_module_name)]+0xcbbd): more undefined references to `VTT for sc_core::sc_signal<bool>' follow
./thermal_IF.o: In function `__static_initialization_and_destruction_0(int, int)':
thermal_IF.cpp:(.text+0x3d): undefined reference to `sc_core::sc_api_version_2_2_0::sc_api_version_2_2_0()'
./NoximVLink.o: In function `__static_initialization_and_destruction_0(int, int)':
NoximVLink.cpp:(.text+0xbd): undefined reference to `sc_core::sc_api_version_2_2_0::sc_api_version_2_2_0()'
./NoximLog.o: In function `__static_initialization_and_destruction_0(int, int)':
NoximLog.cpp:(.text+0x3d): undefined reference to `sc_core::sc_api_version_2_2_0::sc_api_version_2_2_0()'
collect2: error: ld returned 1 exit status  ""
Thank you in advance.

Ayaz-Hussain, 2015-10-01:
Respected Sir i am also getting the same kind of problem in installation of the access noxim .Kindly help me to resolve this issue.

smonteleone, 2015-10-05:
Hi, sadly this problem depends on the fact that Access Noxim has been developed for an old version of Noxim. While waiting for an updated version of Access Noxim (not developed by us) you could try using the old Noxim version available at https://github.com/davidepatti/noxim/archive/noxim-2010.zip

closed 2015-10-05 by smonteleone

SyedAbubakar, 2015-10-05:
can you at least tell me which version of linux and gnu will be compatible
with systemc-2.2.0.

smonteleone, 2015-10-05:
You can see a list at https://github.com/systemc/systemc-2.2.0, but I think also your version should work.