davidepatti/noxim #3: How to add broadcast support for Router?

victorisildur, 2015-10-14:
Hi, this is a great tool, I'm working on it to simulate a broadcast strategy.

The problem is Router::route() only return one direction. How could broadcast be down in that?

Any thoughts would be helpful.

davidepatti, 2015-10-14:
Looking at route() code, actually a vector of directions is returned:

vector < int >candidate_channels = routingFunction(route_data);

then is applied a selection that pick up only one of them

return selectionFunction(candidate_channels, route_data);

So as first step you should write your own routing function (see examples in routingAlgorithms folder) and NOT apply any selection.
But after that, you should also add some code to the Router::txProcess() in order to detect and manage broadcast situations, since the noxim code assume only one output direction. For example, with broadcast, a packet could reserve more than one output in the reservation table.

victorisildur, 2015-10-14:
Thank you for the explanation.
Yes, for now reservation table is an one-to-one table.
I may have to change the reservation table and see if it works.

victorisildur, 2015-10-14:
Hi! @davidepatti, ABP error occurs every time after modifying reservationTable.

I've modified the ReserveTable so that `rtable.getOutputPort()` return a vector of output_port.

With this modification, in `Router::txProcess()` method's forwarding phase, multiple output_port will be retrieved from the reserveTable, and call `filt_tx[o].write(flit)` multiple times.

However, after all this modification and compile successfully, condition `current_level_tx[o] == ack_tx[o].read()` fails every time in `Router::txProcess()`.

I'm new to SystemC, could you please give me some hints.

closed 2015-10-16 by victorisildur