davidepatti/noxim #4: An ABP error occurs every time after add multiple outputs.

victorisildur, 2015-10-15:
Hi, I've working on supporting broadcast on Noxim, and `ReservationTable` , `Router::txProcess()` , `Router::route()` method are changed so that one input can be reserved for multiple outputs.

However, in the `txProcess()` forwarding phase, where one filt is forwarded to multiple output ports, an Error `cannot forward Input[m] to Output[n]` always occurs for unicast flit after broadcast flit.

A read the NoC.cpp and ABP protocol, I think it'll be allright if I invert `req_tx[o]` and `current_tx[o]` after sending out any filts to `output[o]`.

It's really confusing and hard to debug. Any help would be grateful.

ps: my modification is here -> [victorisildur_forked_noxim](https://github.com/victorisildur/noxim/blob/master/src/Router.cpp)

davidepatti, 2015-10-15:
Maybe reservation table entries should be released too after broadcast is done. 
Sorry but I currently do not have enough time to further investigate the matter. 
Another thing that could help you it's looking to my nanoxim project, derived from noxim. If you look at the Router in nanoxim,  packets of type FIRST_SEGMENT_REQUEST are broadcasted over the network. Thus, since nanoxim router it's based on noxim router, you could get hints on how to those type of packets are managed in order to implement your broadcast...

davidepatti, 2015-10-15:
... I refer in particular to the ACTION_FLOOD mechanism in the TRouter.cpp of nanoxim.

victorisildur, 2015-10-16:
In the end I found the problem is about when to pop the buffer. 
We must carefully pop the buffer because in broadcast some output maybe blocked while others are not, so we can only pop the buffer after all outputs are done.
BTW, the noxim's way of releasing ReservationTable seems right, but nanoxim's way isn't.

closed 2015-10-16 by victorisildur