davidepatti/noxim #5: Changing mesh_dim_x mesh_dim_y results in Hub port not bound error

victorisildur, 2015-10-21:
Hi, I tried to change mesh_dim_x and mesh_dim_y to 8*8, 16*4, 4*4 etc.

And I got this error: 
`Error: (E109) complete binding failed: port not bound: port 'NoC.Hub_15.port_97' (sc_in)
In file: ../../../../src/sysc/communication/sc_port.cpp:231`

I'm using mesh topology and don't use wireless hub(it seems like that cuz I didn't configure any thing about wireless). I'm using the `/config_examples/other/config16x16.yaml`

Could you please tell me how to make it right? Thx

davidepatti, 2015-10-21:
In order to reproduce the issue, please post the command line used and the contente of the yaml file.

victorisildur, 2015-10-23:
My commnad line is: `./noxim -config ../config_examples/other/config16x16.yaml`
And the content of `config16x16.yaml` is [config16x16.yaml](https://github.com/victorisildur/noxim/blob/master/config_examples/other/config16x16.yaml)
where path is a routing I implemented, and it only works on 16x16 mesh, not any other.

davidepatti, 2015-10-24:
Using your yaml config with XY works on my noxim, so it's not a yaml issue. 
Try to use XY in your config yaml:
- if it works, then something is wrong in your algorithm (PATH). But this would be strange, since no systemc signal cabling is usually present in the routing algorithm.
- if it DOES NOT work, it should be something you changed in systemc cabling in your noxim fork.

vnd2586, 2016-04-04:
Dear victorislidur,
My system can run it in normal.
./noxim -config ../config_examples/other/config16x16.yaml 
I have not gotten error like your?

closed 2016-05-22 by victorisildur