davidepatti/noxim #6: Hardware Module Diagram

SyedAbubakar, 2015-11-10:
Greetings!
Thanks for such a great and helpful simulator for NOC researchers. I need your help in providing the module diagram of hardware than you used to design simulator. Because it would help users in better understanding of simulator. I know its a huge favor but please provide us. Because my professor wants me to give sketch of simulator in hardware. And as a beginner in SystemC platform I am unable to find that.
Thanks in Advance.

SyedAbubakar, 2015-11-16:
Please give response.
Thanks!

davidepatti, 2015-11-16:
Hi,
you should be more clear about what you mean by hardware diagram. As regards official documentation/publications, at the moment on this short paper is available:
http://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=7245728
...but we are in the process of publishing a long full-detailed paper covering all aspects of noxim.
As you can understand, we cannot release the paper until it has been published.
However, with a discrete knowledge of SystemC, the code should be self-explanatory.
If you have any specific question let me know.

SyedAbubakar, 2015-11-16:
I need the internal structure of router. How it works and other components.
As beginer in SystemC it is difficult for us to understand code because
noxim use advance system level coding. So i am asking for NOC architecture.
If you can give any clue or closest stuff will be well appreciated.
Thank you very much.

davidepatti, 2015-11-16:
For each hardware module (e.g. Router, PE, Hub) Noxim uses a SystemC description of the I/O signals of the module and the processes that are triggered by these signals.
This description can have *multiple* low level hardware implementations. If you don't understand SystemC code it's really hard to figure out how things could work in hardware. Sorry,  but at the momento the only suggestion I can give you is becoming more skilled in SystemC and then looking at each Noxim SC_MODULE, corresponding to the hardware modules. In particular, for each module try to locate:
- I/O signals of the module
- the processes that are sensitive to those signals (specified in the module class constructor).

SyedAbubakar, 2015-11-16:
Thank you. I hope it will solve my problem.

closed 2015-11-16 by davidepatti