davidepatti/noxim #10: Does noxim support adaptive routing ?

AndrewGigie, 2016-05-17:
When i use the command as mentioned in Manual. It says invalid usage.

davidepatti, 2016-05-23:
which command exactly ?

AndrewGigie, 2016-05-23:
-routing fullyadaptive
The fully adaptive routing algorithm.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DDB4FAA8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

davidepatti, 2016-05-23:
Manual is related to old/deprecated version. Anyway,  fullyadaptive was not deadlock free and will not be ported on current version. Please refer to ./noxim -help and the default_config.yaml in config_examples

vnd2586, 2016-07-20:
Hello prof. Davidepatti et.al,
Noxim can only use for topo mesh? is it right?
many thanks!

davidepatti, 2017-03-16:
If you don't change the code (e.g. signals mapping when creating the NoC), the mesh is assumed as deafult. However, signals/port mapping is done dynamically, so you could implement your own topologies (changing also routing of course).
Anyways, even leaving the code as it is, the usage of radiohub allows a small-world style network.

closed 2017-03-16 by davidepatti