davidepatti/noxim #16: Verbosity Option not working?

mvxt, 2017-01-22:
Hello,

I was working with the program and trying to understand the difference between the DETAILED and VERBOSE options. The detailed option works great, outputs a lot of data.

But when I set different verbosity levels, there doesn't seem to be any change in output. Has this option been deprecated?

```
[13:05:31] mikeyvxt:bin git:(master*) $ ./noxim -config default_config.yaml -verbose 3

        SystemC 2.3.1-Accellera --- May  4 2016 15:46:52
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

Loading configuration from file "default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles... 
 done! Now running for 10000 cycles...
Noxim simulation completed.
 ( 11000 cycles executed)
% Total received packets: 2231
% Total received flits: 17855
% Received/Ideal flits Ratio: 0.991944
% Average wireless utilization: 0
% Global average delay (cycles): 14.5177
% Max delay (cycles): 100
% Network throughput (flits/cycle): 1.98389
% Average IP throughput (flits/cycle/IP): 0.0793556
% Total energy (J): 3.35602e-06
% 	Dynamic energy (J): 2.69689e-07
% 	Static energy (J): 3.08633e-06
[13:05:36] mikeyvxt:bin git:(master*) $ ./noxim -config default_config.yaml -verbose 1

        SystemC 2.3.1-Accellera --- May  4 2016 15:46:52
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

Loading configuration from file "default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles... 
 done! Now running for 10000 cycles...
Noxim simulation completed.
 ( 11000 cycles executed)
% Total received packets: 2303
% Total received flits: 18413
% Received/Ideal flits Ratio: 1.02294
% Average wireless utilization: 0
% Global average delay (cycles): 14.307
% Max delay (cycles): 83
% Network throughput (flits/cycle): 2.04589
% Average IP throughput (flits/cycle/IP): 0.0818356
% Total energy (J): 3.35523e-06
% 	Dynamic energy (J): 2.689e-07
% 	Static energy (J): 3.08633e-06
[13:08:35] mikeyvxt:bin git:(master*) $ 
```

davidepatti, 2017-01-22:
yes, for some reasons has not been used as initially designed.
As you already found, -detailed gives you an insight on output statistics.
But you if are interested in what's really going happening

1) enable debug mode by replacing in Makefile:
DEBUG    := # -g -DDEBUG
with 
DEBUG    := -g -DDEBUG
2) make clean
3) make

Since the amount of simulation log is quite big, redirection of output is strongly suggested.
e.g.

./noxim <YOUR_OPTIONS_HERE> output_log.txt

RadFarhad, 2017-03-04:
Hi.
with change DEBUG line in ./bin/Makefile such as 1,2 and 3 step witen by Dr.davidepatti, you will see different verbosity levels changes.

closed 2017-03-16 by davidepatti