davidepatti/noxim #17: Noxim error "No YAML configuration file found! Use -config to load examples from config_examples folder"

MostafaSaied, 2017-01-28:
am-0107907988@ubuntu:~/Desktop/noxim/bin$ ./noxim -sim 40000 -warmup 5000 -dimx 8 -dimy 8 -buffer 2

        SystemC 2.3.1-Accellera --- Jan 28 2017 09:11:03
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

No YAML configuration file found!
Use -config to load examples from config_examples folder

RadFarhad, 2017-02-24:
Hi.
in "~/noxim/bin$"  command line  write:
./noxim -config ../config_examples/default_config.yaml

MostafaSaied, 2017-02-24:
Thanks a lot,
Done!

closed 2017-02-24 by davidepatti

yadavashish1397, 2020-05-05:
Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
ERROR: Cannot read param verbose_mode.

Please suggest the solution of this error