davidepatti/noxim #19: Depreciation of std::gets

tblah, 2017-03-13:
When compiling noxim using the included script, I encountered an error because GCC could not find std::gets (which is used included from systemc.h). 

It seems that std::gets is no longer part of C++ since C++14 (see http://en.cppreference.com/w/cpp/io/c/gets), so I presume that my gcc version (6.2.0-5ubuntu12) was defaulting to this newer standard. 

I was able to fix the build by forcing gcc to use C++11 (which included std::gets). I did this by modifying noxim's Makefile OPTS variable so as to use the option --std=c++11 in addition to what was already there. I have not checked whether versions of systemc newer than the one used by the installation script fix this issue.

davidepatti, 2017-03-14:
Thank you for reporting this issue.
We are investigating if it is more convenient to set the c++11 flag as default or just put a comment in the makefile.

davidepatti, 2017-07-04:
Forced C+11 standard as default.

closed 2017-07-04 by davidepatti