davidepatti/noxim #21: Error: (E549) uncaught exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa

juanjosegcc, 2017-04-07:
I've installed noxim in my Ubuntu 16.04 machine, with libyaml-cpp-dev and libboost-dev packages from official Ubuntu repositories.

Executing the command:
```
[...]/noxim/bin$ ./noxim -config ../config_examples/default_config.yaml 

        SystemC 2.3.1-Accellera --- Apr  7 2017 12:26:23
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

		Noxim - the NoC Simulator
		(C) University of Catania

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

Error: (E549) uncaught exception: yaml-cpp: error at line 0, column 0: invalid node; 
this may result from using a map iterator as a sequence iterator, or vice-versa
In file: ../../../../src/sysc/kernel/sc_except.cpp:100
```

Have I done something wrong? I just followed the install instructions in the doc/INSTALL.txt file...

juanjosegcc, 2017-04-07:
Fixed, the way I did it was the following:

Instead of installing the libyaml-cpp-dev from Ubuntu official repositories, I downloaded the latest version from https://github.com/jbeder/yaml-cpp/archive/master.zip, as stated in the doc/INSTALL.txt (alternative way).

Once yaml-cpp was built and installed in a custom folder of mine, I modified the noxim Makefile to read:

```
##### LIBRARIES CONFIGURATION #####

SYSTEMC := /home/juanjosegcc/local/systemc/2.3.1a
YAML    := /home/juanjosegcc/local/yaml-cpp/0.5.3

##### DIRECTORIES #####

SYSTEMC_LIBS := $(wildcard $(SYSTEMC)/lib-*)

SRCDIR  := ../src
SUBDIRS := $(filter %/,$(wildcard $(SRCDIR)/*/))
OBJDIR  := ./build

SRCS := $(wildcard $(SRCDIR)/*.cpp) $(wildcard $(addsuffix *.cpp,$(SUBDIRS)))
OBJS := $(subst $(SRCDIR),$(OBJDIR),$(SRCS:.cpp=.o))

MODULE := noxim

##### COMPILER OPTIONS #####

CXX      := g++
OPT      := -O3
DEBUG    := # -g -DDEBUG
OTHER    := -Wall -DSC_NO_WRITE_CHECK -std=c++11# -Wno-deprecated
CXXFLAGS := $(OPT) $(OTHER) $(DEBUG)

INCDIR := -I$(SRCDIR) -isystem $(SYSTEMC)/include -I$(YAML)/include
LIBDIR := -L$(SRCDIR) -L$(SYSTEMC_LIBS) -L$(YAML)/lib

LIBS := -lsystemc -lm -lyaml-cpp 

SPACE := $(subst ,, )
VPATH := $(SRCDIR):$(subst $(SPACE),:,$(SUBDIRS))

##### RULES #####

$(MODULE): $(OBJS)
	$(CXX) $(LIBDIR) $^ $(LIBS) -o $@ 2>&1 | c++filt

$(OBJDIR)/%.o: %.cpp
	@mkdir -p $(dir $@)
	$(CXX) $(CXXFLAGS) $(INCDIR) -c $< -o $@

clean:
	rm -f $(OBJS) $(MODULE) *~

depend: 
	makedepend $(SRCS) -Y -f- 2>/dev/null | sed 's_$(SRCDIR)_$(OBJDIR)_' > Makefile.deps

-include Makefile.deps
```

As you can see the variables SYSTEMC and YAML at the beginning were modified with my custom folders (just as the doc/INSTALL.txt suggests), but also in variable OTHER you should add the compiler option -std=c++11 in order to build with the yaml-cpp library.

Thats it! Looks like the latests version in Ubuntu repositories of yaml-cpp is v.0.5.2, but the version downloaded from the author's repository is v0.5.3.

Kind regards

closed 2017-04-07 by juanjosegcc

Jiji10, 2017-11-21:
Hello, 
I have the same problem, when I execute Noxim, I have the following error:
                                ./noxim -config ../config_examples/default_config.yaml

    SystemC 2.3.1-Accellera --- Nov 20 2017 10:56:41
    Copyright (c) 1996-2014 by all Contributors,
    ALL RIGHTS RESERVED

	Noxim - the WNoC Simulator
	(C) University of Catania

	Adaptations by LGGM - UdeA

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

Error: (E549) uncaught exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
In file: ../../../../src/sysc/kernel/sc_except.cpp:100

The noxim Makefile is as follow:
##### LIBRARIES CONFIGURATION #####

SYSTEMC := /usr/local/systemc-2.3.1
YAML := /usr/local/include/yaml-cpp

##### DIRECTORIES #####

SYSTEMC_LIBS := $(wildcard $(SYSTEMC)/lib-*)

SRCDIR  := ../src
SUBDIRS := $(filter %/,$(wildcard $(SRCDIR)/*/))
OBJDIR  := ./build

SRCS := $(wildcard $(SRCDIR)/*.cpp) $(wildcard $(addsuffix *.cpp,$(SUBDIRS)))
OBJS := $(subst $(SRCDIR),$(OBJDIR),$(SRCS:.cpp=.o))
DEPS := $(OBJS:.o=.d)

MODULE := noxim

##### COMPILER OPTIONS #####

CXX      := g++
OPT      := -O3 -MMD
DEBUG    := #-g -DDEBUG
#OTHER    := -Wall# -Wno-deprecated
OTHER    := -Wall -DSC_NO_WRITE_CHECK -std=c++11# -Wno-deprecated

## *************
## Added by LGGM
## *************

## Change the following options to obtain different simulation information
## You might need to perform a make clean after changing any option

# if DEFINE_WNOC_ASSERT is defined, Noxim stops when abnormal situations occur.
WNOC_SIM_ASSERT := -DDEFINE_WNOC_ASSERT
# if DEFINE_WNOC_LOG is defined, Noxim writes simulation execution logs to the 
# specified log file about WNoC.
WNOC_SIM_LOG := -DDEFINE_WNOC_LOG
WNOC_SIM_OPS := $(WNOC_SIM_ASSERT) $(WNOC_SIM_LOG)

# if DEFINE_TM_ASSERT is defined, Noxim stops when abnormal situations occur.
TM_SIM_ASSERT := -DDEFINE_TM_ASSERT
# if DEFINE_TM_WARNING is defined, Noxim shows warnings when they occur.
TM_SIM_WARNING := # -DDEFINE_TM_WARNING
# if DEFINE_TM_SIMCONFIGLOG is defined, tm_engine writes simulation configuration 
# logs to the specified log file or the standard output.
TM_SIM_CONFIGLOG := # -DDEFINE_TM_SIMCONFIGLOG
# if DEFINE_TM_SIMEXECLOG is defined, Noxim writes simulation execution logs to the 
# specified log file.
# if DEFINE_TM_SIMEXECLOG_EXTENDED is defined, Noxim writes extended logs.
TM_SIM_SIMEXECLOG := # -DDEFINE_TM_SIMEXECLOG -DDEFINE_TM_SIMEXECLOG_EXTENDED
# if DEFINE_TM_SIMSTATSLOG is defined, tm_engine writes simulation statistics logs 
# to the specified log file or the standard output.
TM_SIM_SIMSTATSLOG := # -DDEFINE_TM_SIMSTATSLOG
# if DEFINE_TM_SIMSTATSSUMMARY is defined, tm_engine writes simulation metric statistics summary 
# to the specified log file or the standard output.
TM_SIM_SIMSTATSSUMMARY := -DDEFINE_TM_SIMSTATSSUMMARY
TM_SIM_OPS := $(TM_SIM_ASSERT) $(TM_SIM_WARNING) $(TM_SIM_CONFIGLOG) $(TM_SIM_SIMEXECLOG) $(TM_SIM_SIMSTATSLOG) $(TM_SIM_SIMSTATSSUMMARY)

CXXFLAGS := $(OPT) $(OTHER) $(DEBUG) $(WNOC_SIM_OPS) $(TM_SIM_OPS)

INCDIR := -I$(SRCDIR) -isystem $(SYSTEMC)/include -I$(YAML)/include
LIBDIR := -L$(SRCDIR) -L$(SYSTEMC_LIBS) -L$(YAML)/lib

LIBS := -lsystemc -lm -lyaml-cpp 

SPACE := $(subst ,, )
VPATH := $(SRCDIR):$(subst $(SPACE),:,$(SUBDIRS))

##### RULES #####

$(MODULE): $(OBJS)
	@echo "Linking: $@ ..."
	@$(CXX) $(LIBDIR) $^ $(LIBS) -o $@ 2>&1 | c++filt
	@echo "Process completed."

-include $(DEPS)
$(OBJDIR)/%.o: %.cpp
	@echo "Compiling: $< ..."
	@mkdir -p $(dir $@)
	@$(CXX) $(CXXFLAGS) $(INCDIR) -c $< -o $@

clean:
	@echo "Removing files ..."
	@rm -rf $(OBJS) $(MODULE) *~ $(OBJDIR)
	@echo "Process completed."