davidepatti/noxim #23: Order not enforced between Router::rxProcess() and Router::txProcess()

jjgcc, 2017-04-28:
# Router.cpp

Model execution can behave differently if rxProcess() is called before txProcess(). This way, txProcess() views the flit stored at *buffer[i]* at the same cycle at which the flit was stored by rxProcess().

This happens because _SystemC_ does not guarantee the execution order of _SC_METHODs_ happening at the same clock cycle. It looks like the definition order at the constructor of the module is taken into consideration, but it does not enforce it necessarily.

It must be enforced the following order of the operations: TX **before** RX.

priya0777, 2017-05-07:
hi can you please help me understanding how to check for a given sender tile wheter the last file is delivered to its immediate neighour?

davidepatti, 2017-05-08:
I'm closing this issue since your pull request has been merge. Thank you for your contribution.

closed 2017-05-08 by davidepatti