davidepatti/noxim #25: Noxim Virtual Channles

sourockg, 2017-05-02:
Hello! 
I am trying to figure out how virtual channels work in Noxim.  Please help me to understand how they are implemented?

jjgcc, 2017-05-02:
For what I've seen so far, Noxim does not implement virtual channels yet. You would have to implement it yourself. To do so, take a look at files Router.h and Router.cpp.

sourockg, 2017-05-02:
Thank you for the information. I will try to implement it. 
Would like to know if you have implemented it before and was it complex task?

jjgcc, 2017-05-02:
It is not a trivial task, but it is not so different from what it is already implemented with "naked" buffers. Just take into account that now in the Router::rxProcess(), flits must be saved into buffer[port][vc], and ACKs should be identified also by its VC, that would need more signals (one per port and VC probably).

Also in the Router::txProcess() you will need a more elaborated reservation_table (with VCs).

To accomplish all this, you could start adding a new field to the Flit struct identifying the VC it belongs to.

sourockg, 2017-05-02:
Yes! I had the notion that it will require more signals and buffer structure will change for VCs. Now I can see clear what needs to be done with your reply. Thank you once again.

jjgcc, 2017-05-02:
You may not need more signals if each Flit and ACK has a field identifying the VC it belongs to.

sourockg, 2017-05-02:
Sure! I will work on it. Lets see how it turns out.

davidepatti, 2017-05-03:
Hi guys, a Noxim co-author speaking here. 
Actually, there was an experimental version including VC, but was mainly developed by a student and never really tested and so it may be considered as deprecated.
So, your effort could be useful. Please let me know to which specific VC model are your referring to ? Can you provide a reference paper so that we can have an idea? Thank you!

sourockg, 2017-05-03:
Hello! Thank you for your reply.
I am working on NoC project using Noxim. This project work will improve with VCs in the Noxim. There is no specific paper for VCs I am refering to currently. After above discussion I got the basic idea how it can be done and looking for ways to implement it. If you can also suggest some papers and give direction that will be great. I look forward to hearing from you. Thank you!

jjgcc, 2017-05-03:
I think what @davidepatti means is about the execution model. Adding VCs to the router architecture may change its execution model. For example: it could add an extra stage (Virtual Channel Allocation, VCA), so router would have a deeper pipeline (one extra stage for VCA). Or this may be performed at the second stage where txProcess() chooses the output port in conjunction with the output VC.

That's (I think) what he means when asking about a reference paper.

Take a look at this paper, (Equation 2), where they propose a 6-stage pipelined router:

Mello, A., Tedesco, L., Calazans, N., & Moraes, F. (2005, September). Virtual channels in networks on chip: implementation and evaluation on hermes NoC. In Proceedings of the 18th annual symposium on Integrated circuits and system design (pp. 178-183). ACM.

sourockg, 2017-05-03:
@jjgcc Thank you for clarification. May be my earlier comment was ambiguous but this (papers about execution model) is exactly what I wanted when I asked for suggestion and direction. There are few papers about VCs I could find today. But needed suggestion which will be best practise to follow. I did see above paper today but didnt go through it. Will refer it now. Your suggestions are very helpful in this process. Keep them coming. Thank you once again.

davidepatti, 2017-05-03:
yes, I meant some reference so to have a common idea on the execution model. I will take at look to the paper tomorrow.
Now, but  I'm talking only from a Noxim-developer "perspective", it could be interesting to have the chance to integrate your contribution in the official version. With this idea in mind, implementing it as a simple additional code in the tx/rx processes would be a good choice. I mean: without re-writing the whole Router, but just adding an additional step that may be executed or not, depending on user enabling/disabling VC.  In short, it would be good that, when VC are disabled, the results would be exactly the same as before. Of course, it's your work after all and you can implement your the way you prefer :)

sourockg, 2017-05-03:
@jjgcc  will you be working on implementation?

jjgcc, 2017-05-03:
Sure! Just post a pull request and I can help you with the implementation.

sourockg, 2017-05-03:
Thats good to know! Will do it soon after I complete prerequisite tasks.

sourockg, 2017-05-12:
Hello  @davidepatti and @jjgcc 
I was building some knowledge base to implement ideas as discussed.  As @davidepatti  mentioned I am working out the idea of VCs which can be activated and deactivated as per user need. 
I am stuck in flit transmission. Noxim uses Alternate Bit Protocol for flit transmission. The paper referred by @jjgcc, Hermes NoC uses credits based flit transmission. There were few mentions on Internet that credit based flit transmission is preferable for VCs. I would like your advice if we can continue with ABP for flit transmission in VCs or should we consider credit based flit transmission.
Thank you!

jjgcc, 2017-05-12:
I would use ABP as it's already implemented and it's going to be faster for you to get things going in the short term, the paper I sent you was to highlight the design of a pipelined router scheme and its execution model.

Later, if you have time, you could add a credit-based flow control as an alternative for ABP. Credit-based flow control is not as simple as ABP, so it will take time to get a proper implementation. Keep in mind that credit-based flow control keeps track of the buffer occupation, while ABP does not, but ABP is simpler and I suggest you to stick with it for now.

sourockg, 2017-05-12:
Yes I agree! I was looking for ways to implement the credit based flow control but that sure look like lot of work. I will make the changes as per my understanding and will send you pull request for verification and if you can modify it more. Thank you once again! :)

jjgcc, 2017-05-12:
I'm glad my answer suits you :D. Unless you want to include preemptive congestion management or similar, ABP is the right choice at this stage.

sourockg, 2017-06-19:
Hi! Because of my Master's Thesis deadline I had to put this work on hold. 
I might start working on VCs after this. If I get it working will let you know. 
Thank you for all help. :)

closed 2017-11-24 by davidepatti