davidepatti/noxim #26: how to detect successfull or unsuccessful  transmission to neighouring tile

priya0777, 2017-05-07:
how to understand relation between ack_tx[i],req_tx[i],current_level_tx[i] for knowing successful transmission

jjgcc, 2017-05-08:
I'll try to do my best:

First of all, you must understand how Arbitrating Bit Protocol (ABP) works:

https://en.wikipedia.org/wiki/Alternating_bit_protocol

At Router::rxProcess(), at each cycle, for each input port, if its buffer (buffer[i]) is not full and the requested level for ABP is different from last level, then the received flit is stored at buffer[i], the current level for the ABP is changed and finally, an ACK is returned back (through the corresponding ack_rx[i] signal) with the ABP level changed.

req_tx[i] is set for each port at txProcess() whenever forwarding a flit, so you let know the receiver on the other side, which ABP bit to expect for the next transmission.

I think that's it.

davidepatti, 2017-05-08:
yes, @jjgcc is right. Anyway @priya0777 , just my curiosity, why are you interested about those low level signal details ?

priya0777, 2017-05-08:
Thanks alot @jjgcc .actually @davidepatti i wanted detection of failure of tiles for neighboring nodes

closed 2017-11-24 by davidepatti