davidepatti/noxim #33: Assertion Error

Jason15798, 2017-07-13:
Hi,
> systemc 2.3.1
> gcc 15.1

I run the command below:

`./noxim -config ../config_examples/default_config.yaml`

and it returns error:
```
        SystemC 2.3.1-Accellera --- Jun 14 2015 16:06:33
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED

                Noxim - the NoC Simulator
                (C) University of Catania

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
noxim: /software/public/systemc/2.3.1/include/sysc/kernel/sc_process.h:616: void sc_core::sc_process_b::reference_increment(): Assertion `m_references_n != 0' failed.
Aborted
```
**It's almost the same as this issue https://github.com/davidepatti/noxim/issues/9, but it doesn't work for me.**

I've been troubled for a few days. Anyone can help me fix it?

davidepatti, 2017-07-14:
- Which ubuntu/Linux version ?
- Did you updated to the latest version of noxim ?

davidepatti, 2017-07-14:
anyway, just to be sure, I installed the latest version of Noxim, using the automatic script on the latest LTS of Ubuntu 16.04.2, everything worked perfectly.

SIRINE7878, 2017-07-14:
Hi,
I run the command below:

./noxim -config ../config_examples/default_config.yaml 
with 10 x 10 Mesh topology

and it returns error:

Loading configuration from file "../config_examples/my_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles... 
 done! Now running for 10000 cycles...
noxim: ../src/routingAlgorithms/../Utils.h:155: Coord id2Coord(int): Assertion `coord.y < GlobalParams::mesh_dim_y' failed.
Abandon (core dumped)

Anyone can help me fix it? and how much the maximum cores can be supported by noxim ?

SIRINE7878, 2017-07-14:
Please can you post the content of the yaml file using 64 x 64 Cores to make a comparison and detect my error.
THANK YOU IN ADVANCE...

davidepatti, 2017-07-14:
just look in the config_examples directory

SIRINE7878, 2017-07-14:
I need the yaml file using 64 x 64 cores because i try to simulate this bigger noc and it returns :

*** buffer overflow detected ***: ./noxim terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f2d4c78f7e5]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7f2d4c83111c]
/lib/x86_64-linux-gnu/libc.so.6(+0x117120)[0x7f2d4c82f120]
/lib/x86_64-linux-gnu/libc.so.6(+0x116689)[0x7f2d4c82e689]
/lib/x86_64-linux-gnu/libc.so.6(_IO_default_xsputn+0x80)[0x7f2d4c7936b0]
/lib/x86_64-linux-gnu/libc.so.6(_IO_vfprintf+0x7bd)[0x7f2d4c76592d]
/lib/x86_64-linux-gnu/libc.so.6(__vsprintf_chk+0x84)[0x7f2d4c82e714]
/lib/x86_64-linux-gnu/libc.so.6(__sprintf_chk+0x7d)[0x7f2d4c82e66d]
./noxim[0x4512be]
./noxim(sc_main+0x3f7)[0x425dc7]
/home/sirine/noxim/bin/libs/systemc-2.3.1/lib-linux64/libsystemc-2.3.1.so(sc_elab_and_sim+0xd4)[0x7f2d4d3ffc84]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f2d4c738830]
./noxim[0x4125d9]
======= Memory map: ========
00400000-004ea000 r-xp 00000000 08:06 2887711                            /home/sirine/noxim/bin/noxim
006e9000-006ea000 r--p 000e9000 08:06 2887711                            /home/sirine/noxim/bin/noxim
006ea000-006eb000 rw-p 000ea000 08:06 2887711                            /home/sirine/noxim/bin/noxim
006eb000-006f2000 rw-p 00000000 00:00 0 
0133c000-0a90c000 rw-p 00000000 00:00 0                                  [heap]
7f2d4bff8000-7f2d4c4fb000 rw-p 00000000 00:00 0 
7f2d4c4fb000-7f2d4c513000 r-xp 00000000 08:06 7740627                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f2d4c513000-7f2d4c712000 ---p 00018000 08:06 7740627                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f2d4c712000-7f2d4c713000 r--p 00017000 08:06 7740627                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f2d4c713000-7f2d4c714000 rw-p 00018000 08:06 7740627                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f2d4c714000-7f2d4c718000 rw-p 00000000 00:00 0 
7f2d4c718000-7f2d4c8d8000 r-xp 00000000 08:06 7740644                    /lib/x86_64-linux-gnu/libc-2.23.so
7f2d4c8d8000-7f2d4cad8000 ---p 001c0000 08:06 7740644                    /lib/x86_64-linux-gnu/libc-2.23.so
7f2d4cad8000-7f2d4cadc000 r--p 001c0000 08:06 7740644                    /lib/x86_64-linux-gnu/libc-2.23.so
7f2d4cadc000-7f2d4cade000 rw-p 001c4000 08:06 7740644                    /lib/x86_64-linux-gnu/libc-2.23.so
7f2d4cade000-7f2d4cae2000 rw-p 00000000 00:00 0 
7f2d4cae2000-7f2d4caf8000 r-xp 00000000 08:06 7737851                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f2d4caf8000-7f2d4ccf7000 ---p 00016000 08:06 7737851                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f2d4ccf7000-7f2d4ccf8000 rw-p 00015000 08:06 7737851                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f2d4ccf8000-7f2d4ce00000 r-xp 00000000 08:06 7740639                    /lib/x86_64-linux-gnu/libm-2.23.so
7f2d4ce00000-7f2d4cfff000 ---p 00108000 08:06 7740639                    /lib/x86_64-linux-gnu/libm-2.23.so
7f2d4cfff000-7f2d4d000000 r--p 00107000 08:06 7740639                    /lib/x86_64-linux-gnu/libm-2.23.so
7f2d4d000000-7f2d4d001000 rw-p 00108000 08:06 7740639                    /lib/x86_64-linux-gnu/libm-2.23.so
7f2d4d001000-7f2d4d173000 r-xp 00000000 08:06 3410049                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f2d4d173000-7f2d4d373000 ---p 00172000 08:06 3410049                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f2d4d373000-7f2d4d37d000 r--p 00172000 08:06 3410049                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f2d4d37d000-7f2d4d37f000 rw-p 0017c000 08:06 3410049                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f2d4d37f000-7f2d4d383000 rw-p 00000000 00:00 0 
7f2d4d383000-7f2d4d520000 r-xp 00000000 08:06 3016882                    /home/sirine/noxim/bin/libs/systemc-2.3.1/lib-linux64/libsystemc-2.3.1.so
7f2d4d520000-7f2d4d720000 ---p 0019d000 08:06 3016882                    /home/sirine/noxim/bin/libs/systemc-2.3.1/lib-linux64/libsystemc-2.3.1.so
7f2d4d720000-7f2d4d729000 r--p 0019d000 08:06 3016882                    /home/sirine/noxim/bin/libs/systemc-2.3.1/lib-linux64/libsystemc-2.3.1.so
7f2d4d729000-7f2d4d72f000 rw-p 001a6000 08:06 3016882                    /home/sirine/noxim/bin/libs/systemc-2.3.1/lib-linux64/libsystemc-2.3.1.so
7f2d4d72f000-7f2d4d731000 rw-p 00000000 00:00 0 
7f2d4d731000-7f2d4d757000 r-xp 00000000 08:06 7740622                    /lib/x86_64-linux-gnu/ld-2.23.so
7f2d4d7b6000-7f2d4d8b7000 rw-p 00000000 00:00 0 
7f2d4d938000-7f2d4d93e000 rw-p 00000000 00:00 0 
7f2d4d953000-7f2d4d956000 rw-p 00000000 00:00 0 
7f2d4d956000-7f2d4d957000 r--p 00025000 08:06 7740622                    /lib/x86_64-linux-gnu/ld-2.23.so
7f2d4d957000-7f2d4d958000 rw-p 00026000 08:06 7740622                    /lib/x86_64-linux-gnu/ld-2.23.so
7f2d4d958000-7f2d4d959000 rw-p 00000000 00:00 0 
7ffed64f1000-7ffed6510000 rwxp 00000000 00:00 0                          [stack]
7ffed6510000-7ffed6512000 rw-p 00000000 00:00 0 
7ffed653a000-7ffed653c000 r--p 00000000 00:00 0                          [vvar]
7ffed653c000-7ffed653e000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Abandon (core dumped)

THANK YOU IN ADVANCE...

davidepatti, 2017-07-14:
maybe it's too much for your memory...
1) try smaller size to find the breakpoint
2) please provide me the yaml that you are using

SIRINE7878, 2017-07-16:
1. The breakpoint found is: 31 x 31 cores
 2. And the content of config 32x32.yaml is 
[test.docx](https://github.com/davidepatti/noxim/files/1150976/test.docx)

Thank you a lot ...

SIRINE7878, 2017-07-20:
1. The breakpoint found is : 31 x 31 cores
   2. And the content of config 32x32.yaml is
   test.docx <https://github.com/davidepatti/noxim/files/1150976/test.docx>
   3. My memory is around 6 GB

Can you help me fix the problem ?
Thank you a lot ...

Jason15798, 2017-07-24:
What I use is centos 6.8, x64

SIRINE7878, 2017-07-25:
There are any errors in my .yaml file ?

closed 2017-11-24 by davidepatti