davidepatti/noxim #36: What is the best traffic pattern to test the routing algorithm?

mamadymadan, 2017-08-11:
I need to a application specific traffic pattern. I want to bench mark my routing algorithm.what is the best choice?
What is apsra traffic pattern?

davidepatti, 2017-08-11:
Please refer to the main noxim paper cited in the homepage to have details/references about routing  algorithms. As for ASPRA, see  https://www.computer.org/csdl/trans/td/2009/03/ttd2009030316-abs.html

mamadymadan, 2017-08-12:
Thank you for your help. I have already read this article but now I need a data set of different applications so I can examine my algorithm.
In fact, I need a data set that can be cited in the articles.
Are there any data set with these specifications?
Thanks for your follow up.

Sissima, 2017-11-03:
You can find in this paper a study mentioning the most used traffic patterns to test NoC routing algorithms.
[NoC routing protocols – objective-based classification](http://www.sciencedirect.com/science/article/pii/S1383762116300303)

closed 2017-11-24 by davidepatti