davidepatti/noxim #42: A question about source code (NoC.h)

setat, 2017-10-02:
Hi,

A question about source code (NoC.h).
Why are "req", "ack", "flit" and "flit_slots" defined as instances of struct sc_signal_NSWEH? 
but "t" is defined as a pointer to Tile?

Thank you.

davidepatti, 2017-10-02:
I don't understand the question...

HassanMortazavi, 2017-10-04:
Hi 
Because of implementation of simulator .   it's logical ,  in this simulator and in the main function you can see an object of NoC will be define . in this object we have M*N ( mesh dimensions ) tiles(objects) .  ( you can find M*N t pointers in the NoC object.   )
Tile ***t;  means list of pointers to a 2D array ( M*N array)
Tile is a tile that makes from a Router and ProcessingElement.
sc_signal_NSWEH is a struct that indicate a input /output signals .   

Regards 
Hassan Mortazavi

closed 2017-11-24 by davidepatti