davidepatti/noxim #47: What is butterfly traffic distribution?

stdoo, 2017-11-03:
Hi, I can't find any literature about butterfly traffic distribution. There's butterfly topology, obviously they are not the same thing. I can't find the code in noxim related to it either.

Sissima, 2017-11-03:
Hi, 

You can find a definition of butterfly trafic on Nirgam simulator and then code it similarly on Noxim. 
It is defined as: 
//////////////////////////////////
int traffic_patterns::trafficButterfly(int tileID, int num_rows, int num_cols)
{

    int nbits = (int)log2ceil((double)(num_rows*num_cols));
    int dnode = 0;
    for (int i=1; i<nbits-1; i++)
        setBit(dnode, i, getBit(tileID, i));
    setBit(dnode, 0, getBit(tileID, nbits-1));
    setBit(dnode, nbits-1, getBit(tileID, 0));
    if (dnode>=num_rows*num_cols)
    {
        dnode=num_rows*num_cols-1;
    }

    return dnode;

}
/////////////////////////////////
This give you the following result in the case of 4x4 mesh 2D: 

-1--5--4--8-
-3--2--6--10-
-5--9--13--12-
-7--11--15--14-
-8--4--5--1-
-10--6--2--3-
-12--13--9--5-
-14--15--11--7-

Hoping help you. 
Regards,

stdoo, 2017-11-06:
Thanks a lot. I plot it out, and it is not like a butterfly.

closed 2017-11-24 by davidepatti