davidepatti/noxim #50: traffic

setat, 2017-11-15:
Hi,
How to run a table_based traffic in Noxim?

HassanMortazavi, 2017-11-18:
Hi 
You can make a txt file on simulator bin/ directory ( for example t.txt )
after that indicate src and dst nodes and with traffic injection load between this 2 nodes in each line of this file. 
for example copy this line into t.txt file:
64 255 0.0005
3  103 0.0007
It means you wanna inject 0.0005 traffic between 64 & 255 nodes on network and 0.0007 between 3 & 103 nodes simultaneously.

after making this file modify end of YAML configuration on this parts:
traffic_distribution: TRAFFIC_TABLE_BASED
traffic_table_filename: "t.txt"

Regards

closed 2017-11-24 by davidepatti

KacperKubara, 2020-04-12:
What does 0.0005 represent? Did you mean 0.05 % percent of the total traffic?

davidepatti, 2021-10-18:
Just modify the traffic table reader so that it reads the pir AND ALSO a value of packet size.
I think shouldn't be complext, just replicate what has been done for the pir of each flow in order to add also the packet size.