davidepatti/noxim #53: Error: (E549) uncaught exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa

Jiji10, 2017-11-21:
Hello,
when I execute Noxim, I have the following error:
                                 ./noxim -config ../config_examples/default_config.yaml

    SystemC 2.3.1-Accellera --- Nov 20 2017 10:56:41
    Copyright (c) 1996-2014 by all Contributors,
    ALL RIGHTS RESERVED

	Noxim - the WNoC Simulator
	(C) University of Catania

	Adaptations by LGGM - UdeA

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

Error: (E549) uncaught exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
In file: ../../../../src/sysc/kernel/sc_except.cpp:100

Have I done something wrong?

davidepatti, 2017-11-24:
Are you using the original config file without any modifications ?

Jiji10, 2017-11-24:
Hi, 
Yes, i juste change the PATH of SystemC and YAML in Noxim Makefile:
SYSTEMC := /usr/local/systemc-2.3.1
YAML := /usr/local/include/yaml-cpp

davidepatti, 2018-02-20:
You probably wrongly typed some option on YAML file.

closed 2018-03-06 by davidepatti

dylstuart, 2019-01-21:
I'm having the same issue, the YAML file has not been changed.