davidepatti/noxim #54: wireless network on chip

David6386, 2017-11-23:
hi 
How come when we configure wireless network on chip at simulator we don't see any improvement in the number of received packets and Global average delay in comparison to network on chip? Does it need to change the source codes or any other parts?

HassanMortazavi, 2017-11-23:
Hi 
In the end of simulation you can see result of your simulation that one of final parameters is Wireless Utilization. If you had some packets that routed via wireless media, this parameter should not be zero. So you can find you have some packets transmitted via wireless hubs. 
You should have a trade of for using wireless hubs by your routing algorithm. if you use wireless media very much, your performance will be decrease .

David6386, 2017-11-24:
Hi Mr HassanMortazavi

David6386, 2017-11-24:
@HassanMortazavi 
hi
Thank you for your immediate reply. I read one of your comments that you mentioned: "for better use of wireless routing (and more flexible) I commented it from Router.cpp and move it to my RoutingAlgorithm" Is it possible to have your RoutingAlgorithm for increasing my wireless network on chip performance?

HassanMortazavi, 2017-11-25:
Your welcome, 
I found one of wireless routing algorithm in my projects but i modified simulator very much, I hope it can still work and you could use it, but feel free to contact me if you have any question.

You can find this routing in attached files. ( i sent utils.h due to i added some required functions there)

This distributed routing algorithm calculate flit hop count via wireless hub path and without wireless hub path and after that decide to go near wireless hub or route with XY routing to achieve destination. ( I use an adjustment factor for this decision due to prohibit congestion near wireless hubs and wireless media )

[NoximWirelessRouting.zip](https://github.com/davidepatti/noxim/files/1502904/NoximWirelessRouting.zip)

Best Regards
Mortazavi

closed 2017-11-27 by David6386