davidepatti/noxim #56: Collection of packet traces

sidhartharout, 2018-02-05:
I want to take the snapshots of the moving packets in the network at a regular interval say in each 100 cycles. The information in the trace for a particular packet should be like the following:

**timestamp       source       destination         Input port        Out port**

Can anyone help me out how can I collect the same. Or atleast which are the source files whom I should look onto for this purpose.

Actually in detailed simulation mode I am not getting these information. In debug mode I am getting the router by router movement of each packet but if I don't know how can I control the debug mode. I mean if I want to increase my observability on the packet movement, and want to dump more information to the debug output file, then how can I do that.

davidepatti, 2018-02-20:
You could simply make a "grep" command on the output of the log.
Let's say you want to follow the header flit from a packet starting from node 1 to 5.
cat log.txt | grep "H0, 1->5"

Anyway, wherever in the code you add:
LOG "this is my message, the value of variable foo is  " << foo;

you get the timestamp, the noxim object that generated the message and value of the variable.

sidhartharout, 2018-02-23:
Thank you very much sir.

I am facing another problem. Whenever I am using Traffic distribution pattern as "TRAFFIC_RANDOM" or any other synthetic pattern mentioned in the yaml config file, every time I simulate, I get a different traffic trace which is understood as well. This is because the generation of packet is random in this case.

But if I am using a "TRAFFIC_TABLE_BASED" traffic distribution file, I am expecting the packet generation should be controlled by the traffic table provided. But I am getting different traffic traces for multiple simulations in this case as well.

Why this is happening? And if so, then how can I change the tool to get the same traffic traces for multiple simulations with a traffic table?

davidepatti, 2018-03-06:
The current implementantion of traffic table is not a cycle-by-cycle trace, but still a statistical characterization of the communications flows.
In other words, the format is:
<source node> <dest node> <pir>

For example:
0 5 0.01
7 4 0.001

sidhartharout, 2018-03-06:
But sir, if I would take the same table based traffic and simulate it twice, then I should get the traces same for both the simulation since I am taking the same table. But I am getting different traffic traces.

davidepatti, 2018-03-06:
You could simply use a prefixed seed (e.g., -seed 0)

closed 2018-03-06 by davidepatti

sidhartharout, 2018-03-06:
Thank you very much for your help Sir. I just checked and it is working.