davidepatti/noxim #58: Simulator not working properly

shubhamnath2, 2018-03-13:
Simulator is only working for only XY routing algorithm and only for random traffic. Showing error:
noxim: ../src/Router.cpp:146: void Router::txProcess(): Assertion `false' failed.
Aborted (core dumped)

asrani, 2018-03-13:
it works perfectly for me ..

shubhamnath2, 2018-03-19:
@asrani Maybe your Simulator is not updated, check it once, cause for my case, the simulator is working for XY routing not for west first, north last, etc.

davidepatti, 2018-03-19:
Hi to all, mine is working. 
Can you tell me how to replicate the problem ? (e.g., which yaml config are you using ?)

shubhamnath2, 2018-03-19:
default_config.yaml

shubhamnath2, 2018-03-19:
Then I replaced the new files with the old one(without vc) and compiled it,
now it's working.
(I am currently working on ubuntu 16.04 and after march update, this
problem arise)

davidepatti, 2018-03-19:
I just took the original default_config.yaml, I replaced XY with WEST_FIRST. It works. Are you sure you are using the latest version on github ?

davidepatti, 2018-03-19:
Ok, I managed to replicate the issue. I'll let you know asap.

davidepatti, 2018-03-19:
The problem should be now solved in the latest github version.
Just do a "git pull" from src directory and recompile, or reinstall.

shubhamnath2, 2018-03-19:
I'll look into it, thank you for the help.

davidepatti, 2018-03-19:
thank you for reporting the issue. 
I forgot to merge to merge the updates from the latest experimental branch to the main tree.
Everything worked on my local copy because I did not pushed to the github.
My fault, thank you again !

shubhamnath2, 2018-03-20:
Thank you for solving the issue, currently i didn't check the correction(don't want to take risk on my system), but i'll look into it.

shubhamnath2, 2018-03-20:
Thank you for solving the issue, currently i didn't check the correction(don't want to take risk on my system), but i'll look into it

closed 2018-03-20 by shubhamnath2