davidepatti/noxim #60: Noxim Simulation Running, but showing no result!!

seyedelyar, 2018-03-23:
I install cygwin64 in my windows 10, then I install systemc-2.3.1 and yaml-cpp-master exactlly from install.txt in doc file. then download noxim-master from github and install and run.
when i runthe command  ./noxim showing this massege:

 SystemC 2.3.1-Accellera --- Mar 23 2018 12:50:57
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED
        --------------------------------------------
                Noxim - the NoC Simulator
                (C) University of Catania
        --------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

No YAML configuration file found!
 Use -config to load examples from config_examples folder

then I run this command:
$ ./noxim -config ../config_examples/default_config.yaml
showing this:

 SystemC 2.3.1-Accellera --- Mar 23 2018 12:50:57
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED
        --------------------------------------------
                Noxim - the NoC Simulator
                (C) University of Catania
        --------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

then when I run this command : 

./noxim -dimx 4 -dimy 4 -buffer 8 -size 2 8 -sim 10000 -warmup 20000
again show  this massage: 

 SystemC 2.3.1-Accellera --- Mar 23 2018 12:50:57
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED
        --------------------------------------------
                Noxim - the NoC Simulator
                (C) University of Catania
        --------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

No YAML configuration file found!
 Use -config to load examples from config_examples folder

can you help me what do I do?

seyedelyar, 2018-03-23:
I copy the defult_config.yaml and paste it on bin directory and rename to config.yaml and run this coomand : 
./noxim -dimx 4 -dimy 4 -buffer 8 -size 2 8 -sim 30000 -warmup 20000

and face with this massage:

SystemC 2.3.1-Accellera --- Mar 23 2018 12:50:57
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED
        --------------------------------------------
                Noxim - the NoC Simulator
                (C) University of Catania
        --------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

but didnot show the result.

please help me what can i do tu fix this problem and show my result

davidepatti, 2018-03-23:
No YAML configuration file found!
Use -config to load examples from config_examples folder

seyedelyar, 2018-03-23:
i use multi_channel.yaml anad I run this command : 

./noxim -dimx 4 -dimy 4 -config ../config_examples/multi_channel.yaml > my_log.txt

and I face with this massage:  

Noxim - the NoC Simulator
		(C) University of Catania
	--------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "../config_examples/multi_channel.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles... 
Error: (E109) complete binding failed: port not bound: port 'NoC.Hub_3.port_33' (sc_in)
In file: ../../../../src/sysc/communication/sc_port.cpp:231

when I use  this coomand

./noxim -config ../config_examples/multi_channel.yaml > my_log.txt

also I didnot get any result and face with this massage: 

Noxim - the NoC Simulator
		(C) University of Catania
	--------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "../config_examples/multi_channel.yaml"... Done
Loading power configurations from file "power.yaml"... Done

how can I show the result ??

davidepatti, 2018-03-23:
The multi_channel.yaml configuration, is an example of a 16x16 mesh, so if specify different (smaller) sizes it won't work, since there are non-existing nodes  connected to hubs

seyedelyar, 2018-03-23:
but I problem is that I cannot show the result. I run this command but I cannot see any result.

./noxim -routing WEST_FIRST -config ../config_examples/default_config.yaml > my_log.txt

when I use old noxim the result show after run a command but now I cant see anything.

please help me how can I seen that?

fafa1971, 2018-03-23:
You have redirected the standard output to a file, can you open my_log.txt with a text editor and see what it contains?

seyedelyar, 2018-03-23:
I open my_log.txt and see the above massage:

Noxim - the NoC Simulator
(C) University of Catania
--------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

davidepatti, 2018-03-23:
Are you using the latest noxim version ?
There was a problem with WEST_FIRST, fixed some days ago.
1) please try to use a XY routing 
2) please DO NOT redirect the output

seyedelyar, 2018-03-23:
yes, I download the latest noxim that updated 4 days ago. I run this command without any changes : 
./noxim -config ../config_examples/default_config.yaml

and face with this massage: 

 SystemC 2.3.1-Accellera --- Mar 18 2018 20:36:47
        Copyright (c) 1996-2014 by all Contributors,
        ALL RIGHTS RESERVED
        --------------------------------------------
                Noxim - the NoC Simulator
                (C) University of Catania
        --------------------------------------------
Catania V., Mineo A., Monteleone S., Palesi M., and Patti D. (2016) Cycle-Accurate Network on Chip Simulation with Noxim. ACM Trans. Model. Comput. Simul. 27, 1, Article 4 (August 2016), 25 pages. DOI: https://doi.org/10.1145/2953878

Loading configuration from file "../config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done

but again I didnot seen any result !

please help me what shuld I do to see the result?

davidepatti, 2018-03-23:
I'm sorry you can't get the "massage" you expected.
Anyway, from noxim you can only expect a "message". 
Indeed, it's NoC simulator, not a SPA simulator.
I'll investigate the issue next week, hoping that in the meanwhile you find a way to scroll down your shell window and find the missing messages.

seyedelyar, 2018-03-24:
I expect to seen the result then I run my code. in the old version of noxim(noxim-20131016) i can run and show the result, but it hasnot a wireless router. I want to simulate a winoc mapping with noxim but when I run this version I cannot see any result.
please help me what shuld I do to see the result?

closed 2018-06-19 by davidepatti