davidepatti/noxim #63: A question about multicast

CLAY2333, 2018-04-16:
Hi there..
       I am a college student from China and I am using noxim to complete my curriculum design. But I don't quite understand how to implement a multicast routing algorithm on noxim. Can you guide me or have a simple example?
                                                                                                                                                Thanks in CLAY

davidepatti, 2018-04-23:
Hi, 
we have no examples, but you could simply add a new field in the TPacket  type, managing it properly in your own routing code.

CLAY2333, 2018-04-23:
Hi,
Thank you for your answer, but I did not find the TPacket field in the source code, I do not know which file it is in. I do not know how to control the target routing table. I hope you can give me some answers, thank you

closed 2018-05-08 by davidepatti