davidepatti/noxim #64: routing table file  example

NavidDorfeshan, 2018-04-30:
hi 
i want to create my routing table file , but i dont know what is format routing file in noxim.
how can i create my routing table file format in noxim? can u give me a example of routing table file plz?

mpalesi, 2018-05-09:
Hi,

The format of the routing table is as follows. I start stating that a channel is expressed as

`id_router -> id_router`

A generic entry of the routing table has four fields:

Field 1: is the id of the current router;
Field 2: is the input link;
Field 3: the id of destination node;
Field 4: the list of admissible output links.

Here is an example of a fragment of routing table:

```
% Node     In    Dest   Outs
 1   1->1      32   1->0, 1->9
 2   3->2      34   2->10
 2   3->2      36   2->10
 2  10->2       3   2->3
```

That is, the first row refers to the flits coming from the local IP
core of router 1 and whose destination is node 32. According to the
table information the flits will be routed through one of the two
admissible output links in the Outs column (either 1->0 or 1->9).

closed 2018-06-04 by davidepatti

Pi-Hsung, 2022-11-16:
> Hi,
>
> The format of the routing table is as follows. I start stating that a channel is expressed as
>
> `id_router -> id_router`
>
> A generic entry of the routing table has four fields:
>
> Field 1: is the [...]

Are you sure the format is right? it will get the ao.size() error in noxim