davidepatti/noxim #65: simulation results change

SheriefHeshmat, 2018-05-14:
dear prof
when i repeat simulation for the same configuration i receive different results for the average delay and max delay is this normal ??

example of the output:
% Total received packets: 6855
% Total received flits: 13712
% Received/Ideal flits Ratio: 0.952222
% Average wireless utilization: 0
% Global average delay (cycles): 8.39344
% Max delay (cycles): 37
% Network throughput (flits/cycle): 1.52356
% Average IP throughput (flits/cycle/IP): 0.0952222
% Total energy (J): 6.86278e-06
% 	Dynamic energy (J): 1.08195e-07
% 	Static energy (J): 6.75459e-06

second simulation output
% Total received packets: 6990
% Total received flits: 13980
% Received/Ideal flits Ratio: 0.970833
% Average wireless utilization: 0
% Global average delay (cycles): 8.40544
% Max delay (cycles): 35
% Network throughput (flits/cycle): 1.55333
% Average IP throughput (flits/cycle/IP): 0.0970833
% Total energy (J): 6.86384e-06
% 	Dynamic energy (J): 1.09253e-07
% 	Static energy (J): 6.75459e-06

and so on whenever i restart simulation for the same configuration

davidepatti, 2018-06-04:
Each simulation uses a different random seed, so that you can get statistically significant results by using the average of multiple simulations. 
This is necessary, since some feature are only probabilistically determined , for example, the packet injection rates. If you need to repeat a specific simulation, use -seed to force a specific random seed.

aylann, 2018-06-04:
Dear prof ,
I want to modify noxim to have this configuration : 
Each radio hub is allowed to transmit data on an assigned (fixed) channel.
However, every radio hub is able to receive data from all the rx radio channels.
Any thoughts would be helpful.

davidepatti, 2018-06-04:
The last message is a different topic. Please open another issue.

closed 2018-06-04 by davidepatti