davidepatti/noxim #66: Can not simulate using Table_Based traffic distribution

SheriefHeshmat, 2018-05-14:
dear prof
whenever i try to use table based traffic distribution i face this error
(NB : i tried inserting the name of the file for traffic_table_filename with the complete directory and without )
Loading configuration from file "/home/sherief/noxim/config_examples/default_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
NoC::buildMesh()
Reset for 1000 cycles...  done! 
 Now running for 10000 cycles...
noxim: ../src/ProcessingElement.cpp:135: bool ProcessingElement::canShot(Packet&): Assertion `false' failed.
Aborted (core dumped)

am i missing a step for table based traffic distribution ?

thank you for you help

davidepatti, 2018-06-06:
can you show the exact command line are you using ? and the traffic table file ?

SheriefHeshmat, 2018-06-11:
well after deep search i have succeeded to make table based traffic distribution , after reading the very powerful paper you were mentioned in it sir (The analyzes of Network-on-Chip Architectures based on NOXIM simulator ).
but i can run it only from the .yaml file 
i wish i can make a simulation from the command line 
for example this command : ./noxim -dimx 8 -dimy 8 -routing XY -traffic table CGApp_MPEG4.cg -pir 0.010 poisson >> XY_RA_out_MPEG4

always gives me a prompt to use .yaml file (No YAML configuration file found!
 Use -config to load examples from config_examples folder)
and this command as well (./noxim -dimx 8 -dimy 8 -routing XY -traffic table CGApp_MPEG4.cg -pir 0.010 poisson).

thanks for your help

davidepatti, 2018-06-19:
the use of -config is mandatory.

SheriefHeshmat, 2018-06-23:
i had succeeded in running the table_based traffic distribution inside config file as mentioned above 
but i can not load some  parameters in the simulation outside config file
for example these commands run properly

./noxim -config /home/sherief/noxim/config_examples/default_config.yaml
./noxim -detailed -config /home/sherief/noxim/config_examples/default_config.yaml
./noxim -show_buf_stats -config /home/sherief/noxim/config_examples/default_config.yaml.

and i can obtain my desired results as i wish but these never run

./noxim -pir 0.02 poisson -config /home/sherief/noxim/config_examples/default_config.yaml
./noxim -seed 100 -config /home/sherief/noxim/config_examples/default_config.yaml 

i hope to know what i am missing so as to continue with my work ,
thanks a lot for your help sir

SheriefHeshmat, 2018-07-30:
Thanks a lot , i am able to run my work and make scripts for more than one simulation...thanks for the help

closed 2018-07-30 by SheriefHeshmat