davidepatti/noxim #67: modification

aylann, 2018-06-05:
Dear Mr, 

I want to make small changes to allow each radio hub to transmit data on an assigned (fixed) channel.
However, every radio hub is able to receive data from all the rx radio channels.

Any thoughts would be helpful.

davidepatti, 2018-06-06:
You should modify the config.yaml in order to set the RX and TX channels accordingly.

aylann, 2018-06-14:
Hi , @davidepatti this is the configuration I wanted while noxim does not allow it.

Hubs:
    0:  
        attached_nodes: [0,1,4,5]
        rx_radio_channels: [0,1,2,3]
        tx_radio_channels: [0]
    1:
        attached_nodes: [2,3,6,7]
        rx_radio_channels: [0,1,2,3]
        tx_radio_channels: [1]

So i'm trying to modify Hub.cpp to allow each radio hub to transmit data only on one fixed channel.
Thank you for your answer .

sujimasi, 2018-06-18:
hi, I want to change the routing algorithm from xy to north_last.I changed that and gave a new file name called today.yaml.  noxim is telling no such file is found,once i tried with make command also. can anyone help me in this regard

davidepatti, 2018-06-19:
can you post your today.yaml ? did you use the -config command line to specify it ?

sujimasi, 2018-06-21:
hi,
Thank u for replying
I will try to plot graph in matlab. and regarding today.yaml  the only
change i had done is,  the routing algorithm to be north_last instead of XY
in multi_channel.yaml. I gave the command ./noxim -config
../config_examples/multi_channel.yaml -routing north_last for this i got
correct output.If i change the routing algorithm inside the program and try
to execute ./noxim -config ../config_examples/today.yaml I am not getting
output.I am getting such a file does not exist.

sujimasi, 2018-07-25:
sir,
i found out that my file today.yaml is working nicely thank you sir

Prince-Fash, 2019-05-20:
Please can anyone help on how can noxim be used for prediction or link with Artificial Neural Network?