davidepatti/noxim #69: MPEG-4 mapping in noxim

SheriefHeshmat, 2018-06-23:
Dear prof

i had a question if i want to map a real benchmark application in my work to test a proposed routing algorithm. if i want to map the MPEG-4 application for example should i just give the file for the TRAFFIC_TABLE_BASED with this configuration only (inspired by the  wonderful paper : The Analyzes of Network-on-Chip Architectures Based on NOXIM Simulator)
% Communication graph of MPEG-4 application
%	Source	targe
	1	5
	5	1
	2	5
	5	2
	5	9
	9	5
	10	5
	5	10
	4	5
	5	4
	3	5
	5	3
	3	6
	6	3
	4	6
	6	4
	5	11
	11	5
	10	7
	7	10
	11	7
	7	11
	7	12
	12	7
	7	8
	8	7

or i need to add extra work for handling the bandwidth between each node and another as some require 0.5 up to 600 kBits/sec as mentioned in many papers

Sangs19, 2020-06-03:
> Dear prof
>
> i had a question if i want to map a real benchmark application in my work to test a proposed routing algorithm. if i want to map the MPEG-4 application for example should i just give the [...]

Hi,
Did you find solution for this? I was also struggling to understand how to map application CG in noxim. Please help me out

SheriefHeshmat, 2020-06-19:
> Hi,
> Did you find solution for this? I was also struggling to understand how to map application CG in noxim. Please help me out

unfortunately i did not find , if you found a tutorial for understanding this issue in development i will be very glad if you share it

Sangs19, 2020-06-24:
> unfortunately i did not find , if you found a tutorial for understanding this issue in development i will be very glad if you share it

Even am at the same level. Trying to find help. Surely if i find anything will share with you, expect the same from you too as we are on same path.