davidepatti/noxim #71: adaptability regarding

sujimasi, 2018-07-04:
dear prof,
I want to create a fault on a link in 4x4 mesh hybrid topology and i want to check whether deadlock has occured in my topology, otherwise i want to create a deadlock in my topology.is there any project available regarded with this. or can i get a small clue regarding this.

thanks in advance
sujatha

davidepatti, 2018-11-09:
it mostly depend on what you want to do after the fault.
However, if you just want to check if deadlock happens, just modify the routing algorithm in order to catch all the transmissions toward the fault link.
For example, if you want to simulate what happens if  link on direction east of node 2  becomes faulty after clock cycle 5000, you could simply add the following at the end of the routing algorithm:
(WARNING it's pseudo code, use the appropriare variables!)

if (sc_time > 5000)
{
       if (local_id == 2)
       {
             if (chosen_dir == DIRECTION_EAST)
                return NOT_VALID // or maybe introduce another constant to manage in a specific way
      }
}
return chosen_dir; // here we return the chosen direction as usual

sujimasi, 2018-11-09:
Thanku so much sir for your valuable information

closed 2018-12-20 by davidepatti