davidepatti/noxim #74: Multi-Channel Transaction

NavonilChatterjee, 2018-11-08:


davidepatti, 2018-11-09:
Dear Navonil,
your question is a bit too much generic....
what problem are you finding when defining two channels ?

NavonilChatterjee, 2018-11-12:
Respected Sir,

I cannot transmit two packet at a time for two different source destination pair which are connected by wireless hubs.

I have used a 8 x 8 NoC. I have wireless hub connected with router 9, 13, 41 and 45. Now I want to transmit packet from 9 --13, and 41--45 at the same time. But I cannot do that. Packets are getting transmitted one after another, that is, first 9 and 13 is transmitted then 41 and 45 is getting transmitted. Why is this happening? The token policy I have used is TOKEN_PACKET. 

Thanking you,
Navonil Chatterjee.

van-namdinh, 2018-11-15:
Dear authors,
Is there any configuration of .yaml file to run noxim with "TABLE_BASED" option?
Many thanks!