davidepatti/noxim #81: communication Tile-Hub-Tile

deividat, 2018-12-18:
Dear Davide,

Has the communication Tile-Hub-Tile, between two Tiles connected to the same Hub, been implemented?

In the Noxim paper, you say that yes, however looking at the source code I did not see it implemented.

Thank you.

davidepatti, 2018-12-20:
The Tile-Hub-Tile communication you need is already implemented as hardware connections (port mapping etc...) , but requires some specific routing strategy to support it.
That is, you can send a packet from A to B where both are connected to a radio-hub R, but the current default routing logic is to use radiohub only when A and B  nodes are connected to different hubs and you want to use wireless transmission.
To implement what you are talking about, I think you should add some logic in Router.cpp,  where the direction is computed (vector < int > Router::routingFunction(const RouteData & route_data), in order to  capture this situation where you are sending to some node which is connected to the SAME radio-hub.
To handle this case, I think you need the definition of a new output direction, something like: DIRECTION_HUB_TO_LOCAL,so that the radio hub can differentiate  this kind of incoming packets from the usual ones coming as result of DIRECTION_HUB.

deividat, 2018-12-26:
Got it!

Thank you very much

closed 2018-12-26 by deividat