davidepatti/noxim #86: How experiments NoC fault-tolerant routing algorithm in the Noxim？

Seeker-jeff, 2019-01-25:
Dear Professor, I want to implement a fault-tolerant routing algorithm for link failure in NoC, like FADyAD, FoN. But I don't know how to set the link failure rate in Noxim and don't know how to detect these faulty links. Can you give me some advice and methods?

van-namdinh, 2019-03-06:
I also care about this issue! Hope you can find a solution for it!

Seeker-jeff, 2019-04-13:
Hello！For these fault-tolerant routing algorithms such as FoN, it is actually link fault-tolerant routing algorithms. In previous questions, it was mentioned that a faulty network needed a corresponding fault-tolerant routing algorithm. We can skip error detection. We use the flag bit to directly mark the faulty link in the network (the W/E/S/N port of the routing node).This method can be implemented in Noxim with an array or by adding a unit to a class. Then, according to the flag of the node, the routing algorithm judges the fault of the node's W/E/S/N direction and selects the fault-free link route. In this way, these routing algorithms of FoN can be implemented. These are some of the suggestions put forward, and I also hope to discuss them with you.

van-namdinh, 2019-11-15:
> irect

Dear Seeker-jeff.
Could you explain and discuss more about the implementation of FoN?
Thanks!

ishakhari, 2020-10-02:
I am also looking for a solution to deactivate a link or a node in Noxim