davidepatti/noxim #89: pleas Noxim help

azizjarrouj, 2019-04-14:
./noxim: error while loading shared libraries: libsystemc-2.3.1.so: cannot open shared object file: No such file or directory

???
what can I do to Run Noxim simulator
thanks for all

SajeedShahriat, 2019-04-17:
Run the command:

 export LD_LIBRARY_PATH=/<your system c installation path>/systemc-2.3.1/lib-linux64/

from the directory where the Noxim executable file (./noxim) is located (i.e in the bin directory):

hope it helps!

Prince-Fash, 2019-05-20:
1. Can anybody explain how Noxim can be coded to predict traffic flow in Wireless Routers. 
2. How noxim cam be link or programme with Artificial Neural Network

?????
what can I do to Run Noxim simulator with ANN for prediction.......

Thank you all.

closed 2019-06-12 by davidepatti