davidepatti/noxim #93: Modification of the Mesh Topology

NavonilChatterjee, 2019-10-02:
I want to modify the topology to connect some extra links in the noxim simulator. For this I have changed the router.cpp, NoC.cpp, NoC.h, DataStructs.h, Main.cpp. GlobalParam.h
In the GlobalParam.h I have changed the number of ports of a router from 4 to 5. I have named the port as extra. In the datastructs.h I have modified the NoP by adding extra condition for the 5th port. In the router.cpp, in configuration function I have added extra conditions to check and eliminate the connects not present. Finally, I have made extra connects in NoC.cpp and NoC.h.

However, when I try to run the noxim its showing segmentation fault with the following error:
Loading configuration from file "../config_examples/MOT_config.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles... 
Error: (E109) complete binding failed: port not bound: port 'NoC.Tile[06][07]_(#62).port_69' (sc_in)

Can you please tell me how to create this extra port in the router?

GMahmoudi, 2019-10-02:
You can use NoximLong (on github). It is a branch of noxim and supports a fifth connection, which is a Long connection.Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/davidepatti/noxim/issues/93?email_source=notifications\u0026email_token=ADMWNLB2U4HCTOO2XFD54NDQMSXJDA5CNFSM4I4XJAN2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HPEZYBQ",
"url": "https://github.com/davidepatti/noxim/issues/93?email_source=notifications\u0026email_token=ADMWNLB2U4HCTOO2XFD54NDQMSXJDA5CNFSM4I4XJAN2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HPEZYBQ",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

davidepatti, 2019-10-03:
It's not that simple, expecially to debug. I suggest you to update to the latest version that supports delta topologies, just to see how differented port mappings can be done to implement very different topologies. Also remember that if you want to add some port you should has to both the router AND the tile can contains the router. Look at how the RadioHub are connected....

NavonilChatterjee, 2019-10-03:
Thank you for the reply. I am trying to update. If you don't mind I can send you the updated code. If you could have a look. Also, I went through the delta topology. But I am really confused about the topology. Can you upload a figure just to know how the connections are updated.

closed 2020-02-17 by davidepatti