davidepatti/noxim #95: Noxim installation Error

ZulqurnainHaneef, 2019-10-17:
Greetings!

I am trying to install noxim and path to systemc and yaml is set as 

SYSTEMC := ../../systemc-2.3.3
YAML    := ../../yaml-cpp

and i am getting these errors 

g++ -L../src -L../../systemc-2.3.3/lib-linux -L../../yaml-cpp/lib build/Buffer.o build/Channel.o build/ConfigurationManager.o build/GlobalParams.o build/GlobalRoutingTable.o build/GlobalStats.o build/GlobalTrafficTable.o build/Hub.o build/Initiator.o build/LocalRoutingTable.o build/Main.o build/MM.o build/NoC.o build/Power.o build/ProcessingElement.o build/ReservationTable.o build/Router.o build/Stats.o build/Target.o build/TokenRing.o build/routingAlgorithms/RoutingAlgorithms.o build/routingAlgorithms/Routing_DELTA.o build/routingAlgorithms/Routing_DYAD.o build/routingAlgorithms/Routing_NEGATIVE_FIRST.o build/routingAlgorithms/Routing_NORTH_LAST.o build/routingAlgorithms/Routing_ODD_EVEN.o build/routingAlgorithms/Routing_TABLE_BASED.o build/routingAlgorithms/Routing_WEST_FIRST.o build/routingAlgorithms/Routing_XY.o build/selectionStrategies/Selection_BUFFER_LEVEL.o build/selectionStrategies/Selection_NOP.o build/selectionStrategies/Selection_RANDOM.o build/selectionStrategies/SelectionStrategies.o -lsystemc -lm -lyaml-cpp  -o noxim 2>&1 | c++filt
build/Buffer.o: In function `_GLOBAL__sub_I__ZN6BufferC2Ev':
Buffer.cpp:(.text.startup+0x40): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
build/Channel.o: In function `_GLOBAL__sub_I__ZN7Channel11b_transportEiRN3tlm19tlm_generic_payloadERN7sc_core7sc_timeE':
Channel.cpp:(.text.startup+0x40): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
build/ConfigurationManager.o: In function `_GLOBAL__sub_I_config':
ConfigurationManager.cpp:(.text.startup+0x3b): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
build/GlobalStats.o: In function `_GLOBAL__sub_I__ZN11GlobalStatsC2EPK3NoC':
GlobalStats.cpp:(.text.startup+0x40): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
build/Hub.o: In function `_GLOBAL__sub_I__ZN3Hub9tile2PortEi':
Hub.cpp:(.text.startup+0x40): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
build/Initiator.o:Initiator.cpp:(.text.startup+0x40): more undefined references to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)' follow
collect2: error: ld returned 1 exit status

can you guide me how to fix these errors ?

with regards

davidepatti, 2019-10-21:
are you using the automatic script on Ubuntu? (suggested option, see homepage)

Leiyulin, 2019-11-12:
please use systemc-2.3.1(along with the install.txt) instead of 2.3.3,this error caused by the version of systemc installed.

van-namdinh, 2019-12-24:
Hi there, as I tried, noxim runs perfectly on ubuntu.
The easiest way is following the guide in README.

closed 2020-02-17 by davidepatti