davidepatti/noxim #97: Questions on routing table

Lujahi, 2019-11-20:
In the other issue #64, you gave a partial example, I am wondering if these tables need to be fully connected. Say for the line...
 1   1->1      32   1->0, 1->9

Will there be another line in the table that is similar to

 9   1->9      32   1->32

or am I misunderstanding?

Also in the routing table is it possible to use routing id's as needed (say I need more routers than nodes is that allowed?) or are these id's tied to the node they are associated with?

van-namdinh, 2019-12-24:
I tried to run noxim with Table_based routing algorithm but It failed.
Is there someone who run it successfully?

lchang256, 2020-04-28:
I had found through GlobalTrafficTable.h and GlobalTrafficTable.cpp that format given in issue#64 isn't correct.  The correct format is the following:

%Source      Destination      PIR     POR       T_on      T_Off      T_period

Source is id of source of traffic
Destination is destination of traffic
PIR is probability of injection rate
POR is probability of retransmission rate
T_on is cycle when traffic starts
T_off is cycle when traffic stops
T_period is how long before cycle repeats

Out of these, Source and Destination are required.
PIR is defaulted to Traffic Injection Rate in config file if not specified.
POR is defaulted to PIR if not specified.
T_on is defaulted to 0 if not specified.
T_off and T_period is defaulted to end of rest to end of simulation time if not specified.

Playing around with these I was able to get some traffic profile that matched what I expected.

van-namdinh, 2020-05-11:
> I had found through GlobalTrafficTable.h and GlobalTrafficTable.cpp that format given in issue#64 isn't correct. The correct format is the following:
>
> %Source Destination PIR POR T_on T_Off [...]

As I understand about the parameter PIR is Packet Injection Rate, 
For example: If PIR = 0.01 that means 1 packet will be injected into the network each 100 cycles
@lchang256 , could you explain more about PIR is probability of injection rate?
Am I wrong somewhere? 
Thanks!

lchang256, 2020-05-11:
Yes, PIR is packet injection rate.  I had a typo in the write up.  POR is probability of re-transmission.

frogxface, 2021-04-19:
> I had found through GlobalTrafficTable.h and GlobalTrafficTable.cpp that format given in issue#64 isn't correct. The correct format is the following:
>
> %Source Destination PIR POR T_on T_Off [...]

Routing tables and traffic tables are different.
The traffic table only contains information about the input node and output node of the packet.
The routing table should also include more specific route information.
They are not the same thing.

closed 2023-12-05 by davidepatti