davidepatti/noxim #101: Simulation Fails with Flit Size of 256 Bits

lchang256, 2020-02-20:
I am running a simple simulation of mesh config with flit size 256 bits.  However, simulation always fails with the following message:
 noxim: ../src/Power.cpp:88: void Power::configuraeRouter(int, int, int, std::_cxx11::string, std::_cxx11::string): Assertion 'GlobalParams::power_configuration.bufferPowerConfig.leakage.find(key)!= GlobalParams::power_configuration.bufferPowerConfig.leakge.end()' failed.

It looks like it's trying to execut NoC.cpp around line 2181:
    t[i][j]->r->power.configureRouter(GlobalParams::flit_size,
		      			      GlobalParams::buffer_depth,
					      GlobalParams::flit_size,
					      string(GlobalParams::routing_algorithm),
					      "default");
					      

That doesn't match the function declaration in Power.cpp:77:

void Power::configureRouter(int link_width,
	int buffer_depth,
	int buffer_item_size,
	string routing_function,
	string selection_function)

What is buffer_item_size supposed to be?

davidepatti, 2020-02-21:
If you look at the power.yaml file in the bin directory, in the default power model the numbers for 256 size are missing, for example in "network interface" and "buffer" sections. This is not a limitation by itself, it's simply that we did not consider such sizes when measuring with those VHDL model. Anyway, you are free to add those missing entries from your own measurements, or just to interpolate the previous values (16,32,64,128) to create the new ones for 256.

lchang256, 2020-02-22:
Thank you.  This fixed my issue.  This can be closed.

closed 2020-02-22 by lchang256

KHariHaraChaitanya, 2023-04-13:
@davidepatti I have a similar issue where I am unable to run simulations with buffer sizes 5 and 6.  How do I extrapolate the ItemSize Leakage Push Front Pop values to add them in power.yaml file?