davidepatti/noxim #106: Table Based Routing Algorithm Doesn't Work

lchang256, 2020-03-03:
Has anyone tried routing table based sim?  

I created a simple mesh with a routing table per the format in GlobalRoutingTable.cpp.

However, simulation always crashes with file not found.

Debug shows the file name and location being parsed correctly by the simulator, but message is always file not found.

Is there a directory where the file is expected?

Thanks

van-namdinh, 2020-03-19:
Hi @Ichang256
Could you share your configuration file to discuss more.
email me to email vndprogramminges@gmail.com
BR!

lchang256, 2020-03-31:
I have figured out the issue.  There were 2 issues:  1. Placement of routing table needs to be in bin directory 2. The routing table format needs to follow  format in issue #64 and not from the code example fields.

closed 2020-03-31 by lchang256