davidepatti/noxim #107: Error while installing Noxim

Sangs19, 2020-03-15:
[image]
Please help to rectify the error. My systemc version is 2.3.3

Sangs19, 2020-03-16:
g++ -L../src -L -L./usr/local/yaml-cpp/lib build/Stats.o build/Router.o build/LocalRoutingTable.o build/ProcessingElement.o build/Initiator.o build/GlobalStats.o build/Power.o build/Main.o build/TokenRing.o build/ConfigurationManager.o build/Channel.o build/MM.o build/GlobalRoutingTable.o build/NoC.o build/Target.o build/ReservationTable.o build/GlobalParams.o build/Buffer.o build/GlobalTrafficTable.o build/Hub.o build/selectionStrategies/Selection_RANDOM.o build/selectionStrategies/Selection_BUFFER_LEVEL.o build/selectionStrategies/Selection_NOP.o build/selectionStrategies/SelectionStrategies.o build/routingAlgorithms/Routing_ODD_EVEN.o build/routingAlgorithms/Routing_NEGATIVE_FIRST.o build/routingAlgorithms/RoutingAlgorithms.o build/routingAlgorithms/Routing_XY.o build/routingAlgorithms/Routing_WEST_FIRST.o build/routingAlgorithms/Routing_DELTA.o build/routingAlgorithms/Routing_TABLE_BASED.o build/routingAlgorithms/Routing_DYAD.o build/routingAlgorithms/Routing_NORTH_LAST.o -lsystemc -lm -lyaml-cpp  -o noxim 2>&1 | c++filt
/usr/bin/ld: cannot find -lsystemc
collect2: error: ld returned 1 exit status

Please look into this error. Getting it after running "make" command

deividat, 2020-03-17:
Possibly the path to the systemc lib is incorrect in the makefile

I also suggest that you use systemc version 2.3.1a

Sangs19, 2020-03-17:
> Possibly the path to the systemc lib is incorrect in the makefile
> 
> I also suggest that you use systemc version 2.3.1a

> Possibly the path to the systemc lib is incorrect in the makefile
> 
> I also suggest that you use systemc version 2.3.1a

"make" command worked with systemc-2.3.1a. But now am getting the following error after running ./noxim

No YAML configuration file found!
 Use -config to load examples from config_examples folder

Sangs19, 2020-03-17:
> Possibly the path to the systemc lib is incorrect in the makefile
> 
> I also suggest that you use systemc version 2.3.1a

Thanks.. Got it.

closed 2020-05-10 by Sangs19