davidepatti/noxim #109: Traffic Hotspot error

KacperKubara, 2020-04-12:
I've set up a traffic in the config.yaml file as TRAFFIC_HOTSPOT. After running the noxim, I get a resulting error:

```
Loading configuration from file "../config_examples/default_config_hotspot.yaml"... Done
Loading power configurations from file "power.yaml"... Done
Reset for 1000 cycles...  done!
 Now running for 10000 cycles...
Invalid traffic distribution: TRAFFIC_HOTSPOT
```

What am I supposed to do? There is hardly any information on how to use this traffic option. Should I create a table for that traffic? If yes, how?

YatinKumarGupta, 2021-02-07:
I am also facing the same issue while simulating for my routing algorithm

AvishkaDon, 2023-05-06:
> I've set up a traffic in the default_config.yaml file as TRAFFIC_HOTSPOT. After running the noxim, I get a resulting error:
> 
> ```
> Loading configuration from file "../config_examples/default_config_hotspot.yaml"... Done
> Loading power configurations from file "power.yaml"... Done
> Reset for 1000 cycles...  done!
>  Now running for 10000 cycles...
> Invalid traffic distribution: TRAFFIC_HOTSPOT
> ```
> 
> What am I supposed to do? There is hardly any information on how to use this traffic option. Should I create a table for that traffic? If yes, how?

did you happen to figure out how to set up a traffic table with hotspots?