davidepatti/noxim #118: Installation error

TravisDai, 2020-12-16:
I use the command:
 bash <(wget -qO- --no-check-certificate https://raw.githubusercontent.com/davidepatti/noxim/master/other/setup/ubuntu.sh)
Then I get the following errors:
from /home/travis/noxim/bin/libs/yaml-cpp/src/convert.cpp:3:
/home/travis/noxim/bin/libs/yaml-cpp/include/yaml-cpp/node/detail/iterator.h: In member function ‘void YAML::detail::iterator_base<V>::increment()’:
/home/travis/noxim/bin/libs/yaml-cpp/include/yaml-cpp/node/detail/iterator.h:48:
54: error: ‘next’ is not a member of ‘boost’
   48 |   void increment() { this->base_reference() = boost::next(this->base()); }

Can anyone tell me what is wrong out there?

TravisDai, 2020-12-16:
Then I try to use another way(manual) to install noxim in Ubuntu, I install systemC and yaml step by step, and in the directory noxim/bin/ by using the command "make", I get the following error:
/usr/bin/ld: build/Main.o:Main.cpp:(.text.startup+0x101): more undefined references to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)' follow
collect2: error: ld returned 1 exit status

I try two times both in my physical machine and docker virtual machine and get the same errors.

LeiHuang0694, 2021-01-07:
> Then I try to use another way(manual) to install noxim in Ubuntu, I install systemC and yaml step by step, and in the directory noxim/bin/ by using the command "make", I get the following error: [...]

I got this error too in Ubuntu 18.04,,, Have you solved this problem?

TravisDai, 2021-01-07:
Not yet, we can discuss later.

davidepatti, 2021-01-19:
Hi, sorry for the late reply. 
Try the noboost version of the installation script.
bash <(wget -qO- --no-check-certificate https://raw.githubusercontent.com/davidepatti/noxim/master/other/setup/ubuntu_noboost.sh)

LeiHuang0694, 2021-01-20:
> Hi, sorry for the late reply.
> Try the noboost version of the installation script.
> bash <(wget -qO- --no-check-certificate https://raw.githubusercontent.com/davidepatti/noxim/master/other/setup/ubuntu_noboost.sh)

Thanks. Also I changed the version of SystemC from 2.3.3 to 2.3.1. and it works.

closed 2021-01-20 by davidepatti