davidepatti/noxim #123: Intergrated with CLion

frogxface, 2021-04-19:
I'm having some problems with Noxim and want to debug using Clion's debugger.
I followed /noxim/other/setup/README_CLion.txt and successfully opened the code from CLion, but running it gives me an error like this
```
No YAML configuration file found!
 Use -config to load examples from config_examples folder
```
I found that line 52 of Main.cpp calls the configuration function in ConfigurationManager.cpp to read the input characters via `configure(arg_num, arg_vet)`. Where `arg_num` represents the number of string arrays and `arg_vet` is a set of pointers to a different array of characters, e.g.
`arg_vet[0]` points to `"./noxim"`
`arg_vet[1]` points to `"-config"`
`arg_vet[2]` points to `"... /config_examples/defult_config.yaml"`
So I added this code above this line.
```
arg_num=5;
char s1[] = "./noxim";
char *tmp1 = s1;
arg_vet[0] = tmp1;
char s2[] = "./-config";
char *tmp2 = s2;
arg_vet[1] = tmp2;
char s3[] = "./config_examples/my_config.yaml";
char *tmp3 = s3;
arg_vet[2] = tmp3;
char s4[] = "-power";
char *tmp4 = s4;
arg_vet[3] = tmp4;
char s5[] = "./bin/power.yaml";
char *tmp5 = s5;
arg_vet[4] = tmp5;
```
After that, I can type in the terminal `./noxim` and some random characters like.
`./noxim a b c d`
It runs normally. However, running the program with CLion will break.
`Signal: SIGSEGV (Segmentation fault)`
What's going on here?

davidepatti, 2021-04-19:
You don't have to change the arg_vet inside the code. You just need to pass the exact same parameters that you would use from the command line. For example, if you usually run this from the bin directory of noxim:
noxim/bin$ ./noxim -config ../config_examples/default_config.yaml

the you need to add -config ../config_examples/default_config.yaml to CLine "Run configuration" (top right options)

frogxface, 2021-04-19:
> You don't have to change the arg_vet inside the code. You just need to pass the exact same parameters that you would use from the command line. For example, if you usually run this from the bin [...]

I did what you said, and after adding the content in the command line in Run configuration it did run Noxim successfully using **default_config.yaml**, thank you very much for your help.

But there are some new problems.
When I add `> my_log.txt` to the Run configuration, I get the following error.
`Error: Invalid option: >`
When I set TRAFFIC to **TRAFFIC_TABLE_BASED** in the config file, I get the following error.
```
Loading configuration from file "... /config_examples/my_config.yaml"... .Done
Loading power configurations from file "... /bin/power.yaml".... .Done

noxim:/home/frog/noxim/src/NoC.cpp:121 void NoC::buildCommon():Assertion 'gttable.load(GlobalParams::traffic_table_filename.c_str( )); failed.

```
And in the terminal the above operations can run successfully, why is that? I hope I can get your reply, thanks.

frogxface, 2021-04-20:
The way I run Noxim with CLion is as follows.
**File|Settings|Tools|External Tools**
Click **+** to add a tool and set it as follows.
**Tools Settings**
**Program**: Select your Noxim file location, mine is as follows.
_**/home/frog/noxim/cmake-build-debug/noxim**_

**Arguments**: Enter what you type on the command line when you run Noxim in the terminal, e.g.
`-config ../config_examples/my_config.yaml -power ../bin/power.yaml`

**Working directory**: Select the directory where your Noxim files are located, mine is as follows.
_**/home/frog/noxim/cmake-build-debug**_

Click **OK** to create the tool.

From **Tools|External Tools|the name of the tool you just created**, use the tool you just created.

I was able to run Noxim successfully in CLion by using the created tool in this way.
But after adding the tool to Before launch in **Run/Debug Configurations**, it gives the same error as before, i.e. the config file is not read. So it is still not possible to use CLion's debugger for debugging.

To run Noxim in this way, if you use the traffic table, you need to put the traffic table file in the _**/noxim/cmake-build-debug/**_ directory, not in the _**/noxim/bin/**_ directory where you run it in the terminal.

Also errors are reported when using the logging feature, e.g. when running in the terminal, the log is viewed with the following command.
`./noxim -config ./config_examples/defult_config.yaml > my_log.txt`
And when using CLion, an error is reported: .
`Error: Invalid option: >`
This doesn't seem too different from running it in the terminal. Does anyone know how to debug Noxim using CLion's debugger? Looking forward to your reply.

frogxface, 2021-04-20:
You can use CLion debugger normally by using **Program arguments** in CLion's **Run/Debug Configurations**.
The way to do this is to type the contents of the input command line here, e.g.
`-config . /config_examples/default_config.yaml -power ... /bin/power.yaml`
Thank you very much!

closed 2021-06-22 by davidepatti