davidepatti/noxim #125: Error loading shared libraries: libsystemc-2.3.1.so

amitrokade47, 2021-06-22:
I get the following error when I try the wget command
`error while loading shared libraries: libsystemc-2.3.1.so: cannot open shared object file: No such file or directory`

I have systemc on my machine
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04

Can you please let me know how to fix it?

davidepatti, 2021-06-22:
Hi,
are you sure your are using the automatic script installation, as suggested in the first page?
I just used it on ubuntu 21.10 and it worked (...also tried on previous versions)

davidepatti, 2021-06-22:
Be sure to remove any pre-existing noxim directory before trying again, I also update the script to avoid a small problem occurrin when repeated installation attempts are made.

amitrokade47, 2021-06-22:
Actually I have systemc-2.3.1 but not systemc-2.3.1.so file? The systemc-2.3.1 I have is in a dir different than `/usr/lib` . 
Does the script install systemc library as well ?

davidepatti, 2021-06-22:
yes, the install script automatically compiles the required systemc libs and also put the path so that the runtime can find it. Use the script, look at the main page, it's a single command line.

amitrokade47, 2021-06-22:
I have used the wget command at the github page, it fails in the end with the same error  `error while loading shared libraries: libsystemc-2.3.1.so: cannot open shared object file: No such file or directory`

davidepatti, 2021-06-22:
what is the content of : noxim/bin/libs/systemc-2.3.1/systemc.conf 
?
that file indicates the path, but it is automatically copied into:
/etc/ld.so.conf.d/noxim_systemc.conf

so you should have both file, with the same content.

amitrokade47, 2021-06-22:
Both the files contain `<MY_NOXIM_DIR_PATH>/noxim/bin/libs/systemc-2.3.1/lib-linux64`
(BTW I am doing all of this on the server, not on my local machine, I dont think that should affect it though)

davidepatti, 2021-06-22:
Can you post the log of your compilation? I need only from the last lines, around something like:
Libraries have been installed in:
   <YOUR_PATH_HERE!!>noxim/bin/libs/systemc-2.3.1/lib-linux64

amitrokade47, 2021-06-23:
g++ -O3 -Wall -DSC_NO_WRITE_CHECK --std=c++11   -I../src -isystem libs/systemc-2.3.1/include -Ilibs/yaml-cpp/include -c ../src/routingAlgorithms/Routing_NORTH_LAST.cpp -o build/routingAlgorithms/Routing_NORTH_LAST.o
g++ -L../src -Llibs/systemc-2.3.1/lib-linux64 -Llibs/yaml-cpp/lib build/Stats.o build/Router.o build/LocalRoutingTable.o build/ProcessingElement.o build/Initiator.o build/GlobalStats.o build/Main.o build/TokenRing.o build/ConfigurationManager.o build/Channel.o build/MM.o build/GlobalRoutingTable.o build/NoC.o build/Power.o build/Target.o build/ReservationTable.o build/GlobalParams.o build/Hub.o build/GlobalTrafficTable.o build/Buffer.o build/selectionStrategies/Selection_RANDOM.o build/selectionStrategies/Selection_BUFFER_LEVEL.o build/selectionStrategies/Selection_NOP.o build/selectionStrategies/SelectionStrategies.o build/routingAlgorithms/Routing_ODD_EVEN.o build/routingAlgorithms/Routing_NEGATIVE_FIRST.o build/routingAlgorithms/Routing_XY.o build/routingAlgorithms/RoutingAlgorithms.o build/routingAlgorithms/Routing_WEST_FIRST.o build/routingAlgorithms/Routing_DELTA.o build/routingAlgorithms/Routing_TABLE_BASED.o build/routingAlgorithms/Routing_DYAD.o build/routingAlgorithms/Routing_NORTH_LAST.o -lsystemc -lm -lyaml-cpp  -o noxim 2>&1 | c++filt
./noxim: error while loading shared libraries: libsystemc-2.3.1.so: cannot open shared object file: No such file or directory

davidepatti, 2021-06-23:
what's the content of your noxim/bin ?

amitrokade47, 2021-06-23:
`build |  libs |  Makefile |  Makefile.deps |  noxim |  power.yaml`

davidepatti, 2021-06-23:
go there, and type "ldd noxim", show the output

amitrokade47, 2021-06-23:
linux-vdso.so.1 (0x00007ffe049f6000)
        libsystemc-2.3.1.so => not found
        libstdc++.so.6 => /u/ar/.linuxbrew/lib/libstdc++.so.6 (0x00007ff4fc55f000)
        libm.so.6 => /u/ar/.linuxbrew/lib/libm.so.6 (0x00007ff4fc464000)
        libgcc_s.so.1 => /u/ar/.linuxbrew/lib/libgcc_s.so.1 (0x00007ff4fc24c000)
        libc.so.6 => /u/ar/.linuxbrew/lib/libc.so.6 (0x00007ff4fc0b1000)
        /u/ar/.linuxbrew/lib/ld.so (0x00007ff4fc927000)

davidepatti, 2021-06-23:
if you confirm that looking at file:
/etc/ld.so.conf.d/noxim_systemc.conf
it contains a path inside which you can find libsystemc-2.3.1.so
then type:
"sudo ldconfig" 
it's strange because it's what the script does.

amitrokade47, 2021-06-23:
I added the path you mentioned  to `LD_LIBRARY_PATH` and it started working.
Thanks!

closed 2021-06-23 by amitrokade47