davidepatti/noxim #132: why xyz is has better performance than westfirst routing?

cgraider, 2021-12-03:
Hello Noxim family ,
I came across strange problem simulating and evaluating routing algorithms.
XY routing receives more packets and more throughput while westfirst has much less , while base on papers westfirst must have better results.
what am i missing?

mpalesi, 2021-12-04:
Hi,

What kind of traffic are you using? Please note that deterministic XY
routing outperforms the other routing algorithms under uniform random
traffic.

Regards,

Mau

cgraider, 2021-12-04:
Dear Mau,
Thank you so much for pointing that out, yes its uniform random, 
since i have your attention here can i ask you few questions?
1a. I need a bit guide toward using SPLASH-2 or LDPC into Noxim.
    based on previous issues here, dr patti has pointed out that somehow traffic behavior must be 
    mimicked by traffic table, is there any sample code to take a look at ?
1b. mainly my problem with real traffic is on what basis i should modify Packet injection rate
2a. To calculate new router area overhead i know i must use hardware synthesizers, is there any 
    code for current architecture (E.g. Verilog) which implement NoP?
2b.Must papers use Synopsys design compiler, which i cannot afford, is there any free alternative 
      which is trusted by journals and easy to work with?

Dr Palesi it is such an honor to talking to you. I have read few of your papers.
Best regards.
Mohammad

mpalesi, 2021-12-05:
Hi Mohammad,

Find my answers inline below

On Sun, Dec 5, 2021 at 12:55 AM cgraider ***@***.***> wrote:

>
> 1a. I need a bit guide toward using SPLASH-2 or LDPC into Noxim.
> based on previous issues here, dr patti has pointed out that somehow
> traffic behavior must be
> mimicked by traffic table, is there any sample code to take a look at ?
>
Unfortunately we cannot provide a traffic table for SPLASH-2 or LDPC
benchmarks. You should use one of the many system-level simulators (es.,
GEM5) to trace the communication messages among cores and then use the
traffic table option in noxim to mimic that traffic. Traffic table is
simply a text file that reports in each line the following three
information: source node id, destination node id, average packet injection
rate

1b. mainly my problem with real traffic is on what basis i should modify
> Packet injection rate
>
You should compute the packet injection rate to be included in the traffic
table by following, for instance, the technique described in Catania V.,
Salvatore M., Palesi M., Patti D. (2018) Packetization of Shared-Memory
Traces for Message Passing Oriented NoC Simulation. In High Performance
Computing. ISC High Performance 2018. Lecture Notes in Computer Science,
vol 10876. Springer, Cham. https://doi.org/10.1007/978-3-319-92040-5_16

2a. To calculate new router area overhead i know i must use hardware
> synthesizers, is there any
> code for current architecture (E.g. Verilog) which implement NoP?
> 2b.Must papers use Synopsys design compiler, which i cannot afford, is
> there any free alternative
> which is trusted by journals and easy to work with?
>
Unfortunately we cannot locate our vhdl implementation of NoP. However, the
NoP scheme it's very simple and should not be difficult implement it from
scratch starting form the description of NoP algorithm reported in  the
paper. Regarding the synthesis tools, unfortunately, to the best of my
knowledge there are not free alternatives to commercial tools.

Regards,

mau

cgraider, 2021-12-05:
Thank you Dr mpalesi. 
I got all the answers. I mark this thread as Closed.

closed 2021-12-05 by cgraider