davidepatti/noxim #133: Static power for Router-to-Router link commented out

kamilkhan52, 2022-01-08:
I don't understand why the body of the function void `Power::leakageRMCLinkRouter2Router()` is commented out. The function is called from the `perCycleUpdate` function in Router.

davidepatti, 2022-01-12:
Can't find that function anywhere in the code...

kamilkhan52, 2022-01-12:
Thank you for the prompt response! Sorry I mistyped the function name. It is actually `Power::leakageLinkRouter2Router()`. 

On a related note, I find that the static power is always large compared to dynamic power for my simulations, not matter what the load is. This is contrary to what many papers report [Figure below from https://ieeexplore.ieee.org/document/7035011]. 
[image]

From Noxim, I get the following results for a 8x8 mesh, 4 VCs, 16 total buffer slots per router, uniform traffic at 0.020 PIR.

[image]

Is this expected?

davidepatti, 2022-01-14:
The noxim energy estimates are intended to be used according to the values specified in the power.yaml, whose values of static power and dynamic energy strictly depend on the technology used and the particular router implementation.  It should be noticed that the ratio between static and dynamic energy is following a trend in which the static leakage is surpassing the dynamic energy, and,  in our case, we just provided the values obtained for our self-made router implementation in which the buffers tend to dominate the contribution. 
Of course, the values we provided are not the "ultimate" values that fit every scenario: you may be using different technology and a different (more optimal) implementation of the router, and in this case, you should replace the numbers in power.yaml according your model. You could use tools such as [1] to get your values if you don't want to implement everything at the HDL level.

Ps. as for the r2rleakage is a negligible contribution that in our estimate was already embedded in the buffer input component. Anyway, you are correct, and if the model values were updated then you should replace them with yours.

[1] C. Sun, C.-H. O. Chen, G. Kurian, L. Wei, J. Miller,
A. Agarwal, L.-S. Peh, and V. Stojanovic, “DSENT - A Tool Connecting Emerging Photonics with Electronics for OptoElectronic Networks-on-Chip Modeling,” in 6th IEEE/ACM International Symposium on Networks on Chip (NoCS). IEEE, 2012, pp. 201–210

closed 2022-07-18 by davidepatti