davidepatti/noxim #136: Non-rotation problem of the fault-tolerant routing algorithm added in the Noxim

taadaal, 2022-03-24:
Hello, I added a fault-tolerant routing algorithm named "GRADIENT" to the Noxim according to the Noxim_Tutorial PDF, the routing algorithm was successfully added, but I couldn't get output through the files in Config_Example. What is the problem with my routing algorithm?
[image]
In other issues, I figured out to add a fault-tolerant routing algorithm, we need a flag bit added to a class, which class exactly? is that necessary in my case?
This is the article I used: "https://ieeexplore.ieee.org/abstract/document/6385394"

taadaal, 2022-03-24:
Thanks for your prompt response, I enabled DEBUG in "makefile" and this is my "my_log.txt".

[image]
[image]

and when I ran the "./noxim -config ../config_examples/default_config.yaml" with GRADIENT: a fault-tolerant routing algorithm, seems no transmission is happening, As I mentioned before the routing was successfully added.
Do I need to make changes in other Noxim files? For example, should I use a faulty node or add a flag bit to a class?
If files need to be changed, in which files should these changes be made?

davidepatti, 2026-05-30:
Hi,

from the symptoms, the algorithm is probably registered correctly, but the routing function is not returning a valid next hop, so head flits cannot reserve an output port and packets do not make progress.

In Noxim a routing algorithm must return a non-empty `vector<int>` containing valid output directions, for example:

- `DIRECTION_NORTH`
- `DIRECTION_EAST`
- `DIRECTION_SOUTH`
- `DIRECTION_WEST`
- or `DIRECTION_LOCAL` only when the destination is the current node

A good first test is to make your `GRADIENT` algorithm behave exactly like `XY` when no faults are present. If that does not produce normal results, the problem is in the routing implementation or registration.

There is no built-in config option in Noxim to “fault a node” in wired or wireless mode. You need to add a fault model yourself, for example:

1. Add a list/map of faulty nodes or faulty links in the global parameters/configuration.
2. In your routing algorithm, before returning candidate directions, check whether the neighbor reached through that direction is faulty.
3. Never return a direction that points outside the mesh or into a faulty node/link.
4. If all candidate directions are invalid, log the condition clearly, because the packet will otherwise be stuck.

The “flag bit” depends on the algorithm. If the bit is information that must travel with the packet/flit across multiple hops, then it should be added to `Packet`/`Flit` in `DataStructs.h` and copied when flits are generated. If it is only local routing state, then it does not need to be stored in the packet; it can be computed inside the routing function.

So, in your case, the first thing I would check is whether `GRADIENT::route(...)` always returns at least one valid Noxim direction for every source/destination pair in a fault-free 4x4 mesh. After that works, add the faulty-node filtering.