davidepatti/noxim #140: buffer depth error

taadaal, 2022-06-19:
Hi,
When I set the buffer_depth: 10 in config_examples.yaml I get this error.
Why I couldn't set the buffer size more than 8? Where is the problem?
[image]

taadaal, 2022-06-27:
Thanks for responding dear professor,
Yes I Try 10x10, and the problem is from "Utils.h" file this part:
[image]
How can I fix this problem? Which part of the Noxim must be changed?

davidepatti, 2022-07-18:
what don't you just put some print messages to debug the values...

taadaal, 2022-07-18:
I didn't understand the point. What are print messages?

theblueskies, 2022-09-01:
@taadaal You can use `cout` to print messages. That'll help you to see the transition of states. From the error you have pasted, it looks like your modification has caused it to exceed `mesh_dim_y` . Meaning if the max value of y dimension is 10, your changes have just gone above 10 and the assertion has failed.

closed 2022-12-07 by davidepatti