davidepatti/noxim #145: Adding a new Routing Algorithm

1534292451, 2022-09-02:
When I want to adding a new Routing Algorithm, acordding to noxim_tutorial.pdf,  there is 6 Steps, but I can not achieve 5th step:Edit power.yaml to put static & dynamic power cost of doing PIZZA. Could you tell me how to get it?

davidepatti, 2022-09-05:
Can you be more precise?

1534292451, 2022-09-13:
> Can you be more precise?

Thank you very much for your prompt reply. I have a few questions. First, could you please describe in more detail how to add new routing algorithm energy after adding .cpp and .h files when deploying a new routing algorithm? Second, can noxim implement the abacus turn model routing algorithm?

davidepatti, 2022-09-19:
You will just put the static and dynamic power of your algorithm implementation, I mean, the logic implementing the routing decision, not the whole router. It depends on the measurements that you can make with a simple  VHDL simulation, or just approximated according to the most similar among the other algorithms already supported. 
Don't know about abacus, but everything can be implemented, it's just a question of effort.

closed 2022-12-07 by davidepatti